MrChromebox/firmware #336: make: *** No rule to make target 'blobs/soc/intel/kbl/nhlt-blobs/rt5663-2ch-48khz-24b.bin'

bam80, 2022-03-11:
I'm trying to run:
```$ ./build-uefi.sh fizz```

Can't find `rt5663-2ch-48khz-24b.bin` blob nowhere.

MrChromebox, 2022-03-11:
right where the makefile expects it: https://github.com/MrChromebox/blobs/tree/master/soc/intel/kabylake/nhlt-blobs

as I previously said, I don't have everything pushed, nor hooked up in the default branch. It will be ready for next release

closed 2022-03-11 by MrChromebox

bam80, 2022-03-11:
Ah, so the path was wrong - for some reason, we have both:
https://github.com/MrChromebox/blobs/tree/master/soc/intel/kabylake
https://github.com/MrChromebox/blobs/tree/master/soc/intel/kbl
The makefile looked in a wrong place.

Thanks

MrChromebox, 2022-03-11:
it depends on what branch you are building with -- again, my blobs repo is not hooked up the default branch

bam80, 2022-03-11:
I'm building with default branch.
I just cloned your blobs repo additionally.
If you think that's bad idea and I would better wait for the next release - well, fair point.
Still I nearly had the build script run successfully.

MrChromebox, 2022-03-11:
the default branch (currently 2021.07.25) and the blobs repo were never used together, so you will definitely run into a lot of paths that don't match up

bam80, 2022-03-11:
> you will definitely run into a lot of paths that don't match up

Hmm, it seems like this was the only one.
At least the build succeeded.
Does the final image contains all the needed?
```
Built google/fizz (Fizz)
FMAP REGION: COREBOOT
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           60992 none
cpu_microcode_blob.bin         0xef40     microcode      313344 none
intel_fit                      0x5b780    raw               176 none
fallback/ramstage              0x5b880    stage          118350 LZMA (278428 decompressed)
config                         0x78740    raw               777 none
revision                       0x78a80    raw               738 none
build_info                     0x78dc0    raw                97 none
fallback/dsdt.aml              0x78e80    raw             17149 none
ecrw                           0x7d1c0    raw            131072 none
ecrw.hash                      0x9d200    raw                32 none
rt8168-macaddress              0x9d280    raw                 0 none
vbt.bin                        0x9d2c0    raw              1180 LZMA (4608 decompressed)
rt5663-2ch-48khz-24b.bin       0x9d7c0    raw               100 none
(empty)                        0x9d880    null             1316 none
fspm.bin                       0x9ddc0    fsp            405504 none
(empty)                        0x100e00   null             4004 none
fsps.bin                       0x101dc0   fsp            147442 LZ4  (188416 decompressed)
fallback/postcar               0x125e00   stage           19496 none
fallback/payload               0x12aa80   simple elf     803943 none
(empty)                        0x1eef40   null          2129572 none
bootblock                      0x3f6e00   bootblock       20384 none
```

If you still think better not to flash it without debug cable, I won't then.

MrChromebox, 2022-03-11:
for Fizz there might have only been one, but for others there are more. If the firmware built then you are good to go.