MrChromebox/firmware #337: build_packages --board=fizz chromeos-ec: ERROR: No EC boards found

bam80, 2022-03-12:
@MrChromebox 
Could you share your procedure to build EC fw?
I'm trying to follow:
https://chromium.googlesource.com/chromiumos/platform/ec/+/HEAD/docs/getting_started_quickly.md
Got stuck on error:
```
* ERROR: chromeos-base/chromeos-ec-0.0.2-r12250::chromiumos failed (compile phase):
 *   No EC boards found.
```
The problem seems similar as reported here:
https://groups.google.com/a/chromium.org/g/chromium-os-dev/c/e-n53AJeuWM

MrChromebox, 2022-03-12:
* git checkout firmware-fizz-10139.B
* rm -rf ./build/fizz
* make BOARD=fizz

bam80, 2022-03-12:
I'm seeing multiple errors such as this:
```
$ make BOARD=fizz
  HOSTCC  util/ectool
util/ectool_keyscan.c: In function ‘cmd_keyscan’:
util/ectool_keyscan.c:207:9: error: ‘strchr’ argument missing terminating nul [-Werror=stringop-overflow=]
  207 |   pos = strchr(kbd_plain_xlate, key);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
util/ectool_keyscan.c:126:28: note: referenced argument declared here
  126 | static const unsigned char kbd_plain_xlate[] = {
      |                            ^~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile.rules:412: build/fizz/util/ectool] Error 1

bam80, 2022-03-12:
Did you issue your commands in the chroot? (I did)

MrChromebox, 2022-03-12:
try `CFLAGS='-std=gnu89' make BOARD=fizz`

> Did you issue your commands in the chroot? (I did)

of course

bam80, 2022-03-12:
> try `CFLAGS='-std=gnu89' make BOARD=fizz`

Doesn't help.
I also tried to build out of chroot as per
https://www.chromium.org/chromium-os/ec-development/getting-started-building-ec-images-quickly/
, the same error.

bam80, 2022-03-13:
I've removed -Werror from Makefile.toolchain.
Now the error is following:
```
(cr) (firmware-fizz-10139.B) bam@Feedme ~/trunk/src/platform/ec $ make BOARD=fizz
  VERSION ec_version.h
  CC      RO/common/version.o
  LD      RO/ec.RO.elf
core/cortex-m/task.c:37:1: warning: type of ‘cec_task’ does not match original declaration [-Wlto-type-mismatch]
   37 | CONFIG_TASK_LIST
      | ^
chip/npcx/cec.c:1170:6: note: type mismatch in parameter 1
 1170 | void cec_task(void)
      |      ^
chip/npcx/cec.c:1170:6: note: ‘cec_task’ was previously declared here
In function ‘svdm_gfu_config’:
lto1: sorry, unimplemented: Thumb-1 hard-float VFP ABI
lto-wrapper: fatal error: /usr/bin/arm-none-eabi-gcc.real returned 1 exit status
compilation terminated.
/usr/x86_64-pc-linux-gnu/arm-none-eabi/binutils-bin/2.36.1/ld.bfd.real: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [Makefile.rules:353: build/fizz/RO/ec.RO.elf] Error 1

MrChromebox, 2022-03-13:
I don't have any suggestions unfortunately, I build in the chroot using the commands I posted earlier and it compiles without issue

bam80, 2022-03-13:
Full list of commands including chroot creation would be helpful.

MrChromebox, 2022-03-13:
> Full list of commands including chroot creation would be helpful.

I created my chroot years ago, set up my EC build environment years ago. I've updated the chroot since but if I were redoing it now I'd follow the same instructions you are

bam80, 2022-03-13:
I understand. So now we have got a situation when the build is not repeatable.
That is something we should avoid.
Building production output in container should solve that.

bam80, 2022-03-13:
If we could reconstruct the way to build EC fw from scratch I could help with composing a Dockerfile, if necessary.

MrChromebox, 2022-03-13:
this is just not something I have any time at all for, and as it's not unique to my fork, should really be lodged as an issue for the upstream project.

Have you tried using the coreboot docker container? Not that I have used or can provide any support, just an idea

closed 2022-03-13 by MrChromebox

bam80, 2022-03-13:
Understood.
I've filed the issues upstream, didn't get any response so far:
https://issuetracker.google.com/issues?q=reporter:butirsky@gmail.com
I also asked on Chromium OS Development Group.

Yes I actually had to build coreboot in docker container as my host system happened doesn't fully suited.
It didn't help for EC building issue, the same error there:
"lto1: sorry, unimplemented: Thumb-1 hard-float VFP ABI".

> make BOARD=fizz

I think the error is somehow related to the options passed to the compiler.
I added `V=1` flag to make to expose mine, for the reference:
```
$ CROSS_COMPILE_arm=/home/bam/tmp/MrChromebox/coreboot/util/crossgcc/xgcc/bin/arm-eabi- HOSTCC=x86_64-linux-gnu-gcc make BOARD=fizz V=1
./util/getversion.sh > build/fizz/ec_version.h
/usr/bin/ccache /home/bam/tmp/MrChromebox/coreboot/util/crossgcc/xgcc/bin/arm-eabi-gcc -DOUTDIR=build/fizz/RO -DCHIP=npcx -DBOARD_TASKFILE=ec.tasklist -DBOARD=fizz -DCORE=cortex-m -DPROJECT=ec -DCHIP_VARIANT=npcx5m6g -DCHIP_FAMILY=npcx5 -DBOARD_FIZZ -DCHIP_NPCX -DCORE_CORTEX_M -DCHIP_VARIANT_NPCX5M6G -DCHIP_FAMILY_NPCX5 -DFINAL_OUTDIR=build/fizz -Iinclude  -Icore/cortex-m/include  -Icore/cortex-m  -Ichip/npcx  -Iboard/fizz  -Icommon  -Ipower  -Itest  -Icts/common  -Icts/  -Iprivate  -Icommon  -Icommon/vboot  -Idriver  -Idriver/temp_sensor  -Idriver/charger  -Idriver/battery  -Idriver/led  -Idriver/bc12  -Idriver/tcpm  -Idriver/ppc  -Ibuild/fizz  -Itest  -I.     -DTEST_ec -DTEST_EC    -DSECTION_IS_RO -DSECTION=RO  -ffreestanding -fno-builtin -nostdinc -nostdlib -Ibuiltin/ -DHAS_TASK_HOOKS -DHAS_TASK_CHIPSET -DHAS_TASK_PDCMD -DHAS_TASK_HOSTCMD -DHAS_TASK_CONSOLE -DHAS_TASK_POWERBTN -DHAS_TASK_PD_C0 -march=armv7e-m -mcpu=cortex-m4 -march=armv7e-m -mcpu=cortex-m4 -mthumb -Os -mno-sched-prolog -mno-unaligned-access -mfpu=fpv4-sp-d16 -mfloat-abi=hard -flto -g  -Wall -Wundef -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -fno-strict-overflow -Wstrict-prototypes -Wdeclaration-after-statement -Wno-pointer-sign  -ffunction-sections -fshort-wchar -fno-delete-null-pointer-checks -fconserve-stack -DCHROMIUM_EC -MMD -MP -MF build/fizz/RO/common/version.o.d -c common/version.c -o build/fizz/RO/common/version.o
/usr/bin/ccache /home/bam/tmp/MrChromebox/coreboot/util/crossgcc/xgcc/bin/arm-eabi-gcc build/fizz/RO/board/fizz/board.o build/fizz/RO/board/fizz/led.o build/fizz/RO/board/fizz/usb_pd_policy.o build/fizz/RO/chip/npcx/adc.o build/fizz/RO/chip/npcx/clock.o build/fizz/RO/chip/npcx/espi.o build/fizz/RO/chip/npcx/fan.o build/fizz/RO/chip/npcx/flash.o build/fizz/RO/chip/npcx/gpio.o build/fizz/RO/chip/npcx/header.o build/fizz/RO/chip/npcx/hwtimer.o build/fizz/RO/chip/npcx/i2c-npcx5.o build/fizz/RO/chip/npcx/i2c.o build/fizz/RO/chip/npcx/lpc.o build/fizz/RO/chip/npcx/pwm.o build/fizz/RO/chip/npcx/system-npcx5.o build/fizz/RO/chip/npcx/system.o build/fizz/RO/chip/npcx/uart.o build/fizz/RO/chip/npcx/watchdog.o build/fizz/RO/common/acpi.o build/fizz/RO/common/adc.o build/fizz/RO/common/button.o build/fizz/RO/common/cbi.o build/fizz/RO/common/charge_manager.o build/fizz/RO/common/chipset.o build/fizz/RO/common/console.o build/fizz/RO/common/console_output.o build/fizz/RO/common/crc8.o build/fizz/RO/common/dptf.o build/fizz/RO/common/ec_features.o build/fizz/RO/common/espi.o build/fizz/RO/common/event_log.o build/fizz/RO/common/extpower_gpio.o build/fizz/RO/common/fan.o build/fizz/RO/common/flash.o build/fizz/RO/common/fmap.o build/fizz/RO/common/gpio.o build/fizz/RO/common/gpio_commands.o build/fizz/RO/common/hooks.o build/fizz/RO/common/host_command.o build/fizz/RO/common/host_command_pd.o build/fizz/RO/common/host_event_commands.o build/fizz/RO/common/i2c_master.o build/fizz/RO/common/keyboard_mkbp.o build/fizz/RO/common/led_common.o build/fizz/RO/common/main.o build/fizz/RO/common/memory_commands.o build/fizz/RO/common/mkbp_event.o build/fizz/RO/common/panic_output.o build/fizz/RO/common/pd_log.o build/fizz/RO/common/port80.o build/fizz/RO/common/power_button.o build/fizz/RO/common/power_button_x86.o build/fizz/RO/common/printf.o build/fizz/RO/common/pwm.o build/fizz/RO/common/queue.o build/fizz/RO/common/queue_policies.o build/fizz/RO/common/rsa.o build/fizz/RO/common/rwsig.o build/fizz/RO/common/sha256.o build/fizz/RO/common/shared_mem.o build/fizz/RO/common/spi_flash_reg.o build/fizz/RO/common/switch.o build/fizz/RO/common/system.o build/fizz/RO/common/temp_sensor.o build/fizz/RO/common/thermal.o build/fizz/RO/common/throttle_ap.o build/fizz/RO/common/timer.o build/fizz/RO/common/uart_buffering.o build/fizz/RO/common/usb_pd_policy.o build/fizz/RO/common/usb_pd_protocol.o build/fizz/RO/common/usb_port_power_dumb.o build/fizz/RO/common/util.o build/fizz/RO/common/vboot/common.o build/fizz/RO/common/vboot/vb21_lib.o build/fizz/RO/common/vboot/vboot.o build/fizz/RO/common/vboot_hash.o build/fizz/RO/common/version.o build/fizz/RO/common/vstore.o build/fizz/RO/common/wireless.o build/fizz/RO/core/cortex-m/cpu.o build/fizz/RO/core/cortex-m/init.o build/fizz/RO/core/cortex-m/ldivmod.o build/fizz/RO/core/cortex-m/mpu.o build/fizz/RO/core/cortex-m/panic.o build/fizz/RO/core/cortex-m/switch.o build/fizz/RO/core/cortex-m/task.o build/fizz/RO/core/cortex-m/uldivmod.o build/fizz/RO/core/cortex-m/watchdog.o build/fizz/RO/driver/tcpm/ps8xxx.o build/fizz/RO/driver/tcpm/tcpci.o build/fizz/RO/driver/temp_sensor/tmp432.o build/fizz/RO/driver/usb_mux.o build/fizz/RO/power/common.o build/fizz/RO/power/intel_x86.o build/fizz/RO/p [... cut ...]
compilation terminated.
/home/bam/tmp/MrChromebox/coreboot/util/crossgcc/xgcc/bin/../lib/gcc/arm-eabi/8.3.0/../../../../arm-eabi/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [Makefile.rules:353: build/fizz/RO/ec.RO.elf] Error 1
```
Related options found:
```
-march=armv7e-m -mcpu=cortex-m4 -mthumb -Os -mno-sched-prolog -mno-unaligned-access -mfpu=fpv4-sp-d16 -mfloat-abi=hard -flto
```
The other possibility is toolchain now configured somehow differently.