MrChromebox/firmware #340: [hatch/nightfury] Accelerometer sensors not working correctly.

mapmot, 2022-04-12:
Screen orientation detection works correctly with Google firmware. With the same kernel and Mr. Chromebox full UEFI firmware, the sensor readings (/sys/bus/iio/devices/iio\:device*/in_accel_*_raw) get stuck at 32767 going from "left up"  to "normal" orientation. Looks like an overflow somewhere, but I have no idea how to debug this further.

mapmot, 2022-04-17:
I flashed back the stock firmware, and the values returned from /sys/bus/iio/devices/iio:device*/in_accel_*_raw are in the range -16382:16383. It looks like a signedness issue. Maybe the modules cros_ec_sensors and cros_ec_accel expect signed value, but the ec firmware returns unsigned value? Any help in tracking the issue further is appreciated!

mapmot, 2022-04-27:
Traced the issue to the DMI BIOS version string being changed in the MrChromebox firmware.

MrChromebox, 2022-04-27:
> Traced the issue to the DMI BIOS version string being changed in the MrChromebox firmware.

can you elaborate on that?

mapmot, 2022-04-28:
linux/drivers/platform/chrome/cros_ec_lpc.c is looking for DMI_MATCH(DMI_BIOS_VERSION, "Google_"), in case of ChromeOS, and DMI_MATCH(DMI_SYS_VENDOR, "GOOGLE"), in case of custom coreboot firmware (such as MrChromebox). Nightfury is one of the few devices that report the vendor as "Google", instead of "GOOGLE". The only other such device that I know of is caroline.

MrChromebox, 2022-04-28:
> Nightfury is one of the few devices that report the vendor as "Google", instead of "GOOGLE". The only other such device that I know of is caroline.

All Skylake/Apollolake and newer CrOS devices use 'Google' for the vendor instead of 'GOOGLE'. I would submit a patch to fix the driver quirk

MrChromebox, 2022-04-28:
well shit, I already did submit a patch for this back in January. Need to ping and see why it didn't get traction

mapmot, 2022-04-29:
Celebrated too soon. Apparently it works only on the first boot after switching from ChromeOS->MrChromebox. Any subsequent boot after switching to MrChromebox, the problem reappears. Tried switching the version string to Google_Nightfury.12672.434.0 and recompiling/reflashing MrChromebox firmware - same result. Now I am suspecting either incorrect initialization of the EC, or problems with the RW EC firmware. How do I extract ecrw form the ChromeOS firmware - flashdescriptor.bin, me.bin, and ec.RW.flat? Tried replacing ec.RW.flat with the ecro from the firmware scrript backup, but this caused all kinds of problems, like not resuming from sleep, in addition to not solving the problem...

mapmot, 2022-04-29:
In absolutely no way is ChromeOS involved in any of this. I am using exactly the same kernel. Booted under MrChromebox/UEFI - accelerometer doesn't work properly. Booted in developer mode after restoring the backup from the MrChromebox firmware script, with the same kernel signed and dd-ed to kernel partition, same OS (Gentoo) - accelerometer works properly. How do you produce ec.RW.flat in your blobs repository?

MrChromebox, 2022-04-29:
> How do you produce ec.RW.flat in your blobs repository?

by compiling it from my forked version of the Chrome-EC repo, using the correct branch and build target for the board.

Need to know the ecrw version used by stock firmware/ChromeOS to compare to my build and see where the issue is coming from

mapmot, 2022-05-02:
This is from Google firmware with correctly working accelerometer:

RO version:    nightfury_v2.0.2849-c41de27e7
RW version:    nightfury_v2.0.2851-66ac38d381
Firmware copy: RW
Build info:    nightfury_v2.0.2851-66ac38d381 2021-06-02 09:21:23 @chromeos-ci-firmware-us-east1-d-x32-1-rbnz
Tool version:  v0.0.1-8e51ac6 2022-05-02 01:12:25 gentoo@nightfury

MrChromebox, 2022-05-03:
resolved with 2022.05.03 UEFI update; closing

closed 2022-05-03 by MrChromebox