MrChromebox/firmware #341: Asus Chromebox 3 no audio output

bam80, 2022-04-16:
Internal sound chip doesn't seem to be detected on Ubuntu, at least no audio output with jack headset connected

MrChromebox, 2022-04-16:
this isn't a firmware issue, it's a driver+config+DSP firmware issue with all KBL ChromeOS devices outside of ChromeOS.

It works fine under Windows using the proper drivers

bam80, 2022-04-16:
Is there more precise description of the problem somewhere?
Does it require proprietary pieces to work? Can we replicate it on Linux?

MrChromebox, 2022-04-16:
> Is there more precise description of the problem somewhere?

maybe on the galliumOS github issue tracker for KBL audio?

> Does it require proprietary pieces to work? Can we replicate it on Linux?

yes, the DSP firmware from ChromeOS, which is easily copied over. The problem, AIUI, is that the Alsa config/topology files in ChromeOS are for an older ABI version than the Linux kernel uses, and converting them to the new ABI is non-trivial. And as the DSP firmware isn't something that Intel has publicly released, projects like SoF aren't doing anything to support the platform

closed 2022-08-24 by MrChromebox