MrChromebox/firmware #342: HP Chromebook G4 [UEFI BLACK SCREEN]

Kraax, 2022-04-21:
hi, I decided to install Linux on a live USB. so I cranked up my chromebook and did the firmware screw end up im the special place (i have it). then typed the chromebox script and flashed UEFI full firmware and as the man I was watching had it flawlessly, I didn't backup the firmware. now I don't even know if my laptop is turned on because it still has a black screen without no backlight etc. and I do not know what to do from what I read, the bios is not supported by my chromebook (unfortunately I read it after flashing) and if it can still be saved somehow, I would ask for help because I am lost.

Kraax, 2022-04-21:
okay, about backlight i was **wrong** because it little glows when turning _on_ and become completly Black when turning _off_
i noticed too some weird sound from the  chromebook its weird because the laptop dosen't have any fan or something it always been quiet so idk what's happening

MrChromebox, 2022-04-21:
> the bios is not supported by my chromebook

huh? Assuming you have the HP Chromebook *11* G4 (KIP), then it is supported

https://wiki.mrchromebox.tech/Unbricking

closed 2022-04-21 by MrChromebox