MrChromebox/firmware #343: Add documentation For the UFEI Shell Commands

S0yf00d-exe, 2022-04-21:
I was trying to change Bios timezone and finally figured it out by typing `help`

  There is many other commands but I am unable to view them 
*cause it cuts the rests of them off.*

(No way to scroll up that I'm aware of)   

  Thankfully `Set` and `Time` are at the bottom of the command list. I was able to firgure it out from there.

Please provide documentaion for the UEFI Shell Commands, Unless there is a way to scroll up and view them.

Thanks.

MrChromebox, 2022-04-21:
the EFI shell commands are standard and can easily found by googling, there's nothing specific to my fork.

eg, https://docstore.mik.ua/manuals/hp-ux/en/5991-1247B/ch04s13.html

S0yf00d-exe, 2022-04-27:
@MrChromebox

*Thanks!*

For some reason I couldn't find them anywhere...