MrChromebox/firmware #344: Pixelbook 2017 touchpad frequently stops working after reboot

jmontleon, 2022-04-22:
Hi, I have a repo over at https://github.com/jmontleon/pixelbook-fedora for running Fedora on the 2017 Pixelbook.

One of the most frequent complaints I get is that the touchpad isn't working or doesn't work after a reboot. I witness this on occasion, though for some it seems to be much more often. Regardless it seems holding the power button until it shuts down or using Refresh+Power gets it working again.

I'm struggling to understand what could be causing this and was wondering, if from a firmware perspective you might have any ideas. 

I remember hitting this even with ChromeOS ages ago and sometimes getting it back mashing the esc key as described here. I don't know if it might just be something about hte touchpad or its firmware...
https://support.google.com/pixelbook/answer/9133994

Thanks in advance for any feedback you might have.

MrChromebox, 2022-04-22:
which touchpad firmware version being used? I don't have a Pixelbook so not much I can do on my end without someone else triaging the issue

jmontleon, 2022-04-22:
Do you know how I can find what the current version I have running is? If not I'll dig around and figure it out.

FWIW, It seems I'm able to reproduce the problem almost 100% since updating to 4.16. I don't think I saw it very much at all with 4.14.

I did some searching since filing this issue and came across https://bugzilla.redhat.com/show_bug.cgi?id=1442699

If I do as suggested in some of the comments, `rmmod i2c_hid_acpi i2c_hid && modprobe i2c_hid_acpi` the mouse comes back.

I created a systemd service to run these and it masks the problem... 
```
[Unit]
Description=Workaround for Pixelbook's touchpa

[Service]
Type=oneshot
ExecStart=rmmod i2c_hid_acpi i2c_hid
ExecStart=modprobe i2c_hid_acpi
RemainAfterExit=yes

[Install]
WantedBy=graphical.target
```

jmontleon, 2022-04-22:
ran `./flashrom -p ec:type=tp -i EC_RW -r -o /tmp/flashrom.log rose.bin` and with hexedit I see: `rose_v2.0.653-dfd8046c6`

jmontleon, 2022-04-22:
Looks like that's the version on the latest recovery image as well, `./opt/google/touch/firmware/rose_v2.0.653-dfd8046c6.bin`

MrChromebox, 2022-04-24:
then perhaps downgrading the firmware to the version offered by my script would resolve the issue? it improves/fixes i2c-hid compatibility for sure

closed 2026-11-04 by MrChromebox