MrChromebox/firmware #348: Bios Password

junzsalez, 2022-04-27:
I am using the UEFI firmware for the Dell Chromebook 11 3120 (Baytrail) and was wondering if there is a way to set a password on the BIOS?

MrChromebox, 2022-04-27:
there is not currently

junzsalez, 2022-04-27:
Would it be possible to add this feature in a future release? There are a few schools using your firmware on EOL Chromebooks to install Chrome OS FLex (to continue receiving updates) and would be great to have the ability to enable a password to prevent students from booting off a USB etc.

MrChromebox, 2022-04-28:
I don't expect to add one anytime soon, it's not a feature in upstream Tianocore, so I'd have to implement it from scratch which is a PITA

mntn-xyz, 2023-01-23:
I was looking for something similar, and I had the thought that maybe the firmware could be built without support for entering setup menus or the EFI shell. This would prevent someone from changing the boot order, at the cost of making it difficult to troubleshoot. Do you happen to know if this is possible and what configuration flags would be needed? Or what documentation I should reference to dig up this information?

mntn-xyz, 2023-01-23:
I dug through the source and I think I found what would be needed to fully disable the boot menu and EFI shell.

For EDKII, it appears that pressing escape to show the boot menu can be disabled by setting `CONFIG_EDK2_BOOT_MANAGER_ESCAPE` to `n`. To disable the EFI shell you would probably need to set `CONFIG_EDK2_HAVE_EFI_SHELL` to `n`. See https://github.com/MrChromebox/coreboot/blob/2022.10.24/payloads/external/edk2/Makefile.

For legacy SeaBIOS, it looks like the menu can be disabled by setting "show-boot-menu" to 0 per https://seabios.org/Runtime_config. I'm not sure how this would be configured if building the firmware on your own, as I am focused on UEFI for my own use case and didn't have time to dig in further.

Would appreciate any feedback on this approach.

MrChromebox, 2023-01-23:
> For EDKII, it appears that pressing escape to show the boot menu can be disabled by setting CONFIG_EDK2_BOOT_MANAGER_ESCAPE to n

no, that just means it will use the default value of F2 instead of ESC

mntn-xyz, 2023-01-23:
Thanks for the correction, much appreciated.

I did some further research, and although it should be possible to create a "kiosk" style firmware without the menu, it is non-trivial (but not impossible). From [this mailing list discussion](https://edk2.groups.io/g/devel/topic/30907022) it appears that you can strip out UiApp to remove the menu entirely, but there's no configuration option for this. For anyone else who wants to do this, it would require modifying the EDKII source used in the build process. The mailing list discussion should at least provide some hints to get started. You would also need to turn off `CONFIG_EDK2_HAVE_EFI_SHELL` as described above.

You should ensure that your system is bootable before flashing a firmware customized in this way. I believe you should be able to use `efibootmgr` to either remove USB boot entirely, or place it after HDD boot to allow for a fallback boot option if all else fails.

I was hoping this would be easier, so I may or may not get to this, but if I do then I'll try to provide an update with more information. Hopefully this serves as a workaround for anyone who needs this.

MrChromebox, 2023-01-23:
how to you intend to boot a USB to install your OS without either a boot menu or shell?

I'd think a better approach would be to change the boot menu key to some unusual combo (SHIFT+J ?) and hide the text prompt, and shorten the timeout to 1s

> Edit: The biggest unanswered question I have is whether or not the firmware will prioritize EFI partitions from USB drives during boot. If so then I assume that a USB drive with an EFI partition could change the EFI boot order since (as I understand it) that information is stored in the EFI partition. If that is the case then further configuration or modification of EDKII may be needed.

upstream EDK does, my fork prioritizes internal over external

mntn-xyz, 2023-01-23:
Thanks once again for the EDK info, will update my instructions above.

For installing an OS, you would boot first with the standard firmware, install the OS, then flash the locked down firmware. Alternately, install the OS to HDD using a separate box and then install the HDD in the locked down box. The first is obviously preferable as you can verify that the OS configuration boots properly under the standard bootloader.

The method you suggested is also workable as long as you aren't concerned that someone may discover your secret hotkey scheme. With USB boot and setup menus disabled completely, a machine sealed in a tamper-evident enclosure should be significantly hardened against attacks that rely on altering the boot process (for example, installing a rootkit on the main OS via USB boot).

jleightcap, 2023-06-21:
This would be a great addition for securing devices in general.

https://wiki.archlinux.org/title/Unified_Extensible_Firmware_Interface/Secure_Boot#Implementing_Secure_Boot

MrChromebox, 2023-06-21:
> This would be a great addition for securing devices in general. In addition to being able to prevent external boot media, Chromebooks seem to have really great support for using [Secure Boot](https://wiki.archlinux.org/title/Unified_Extensible_Firmware_Interface/Secure_Boot#Implementing_Secure_Boot), but:
> 
> > There are certain conditions making for an ideal setup of Secure boot:
> > 
> > 1. UEFI considered mostly trusted (despite having some well known [criticisms](https://en.wikipedia.org/wiki/Unified_Extensible_Firmware_Interface#Criticism) and vulnerabilities[[1]](https://www.uefi.org/sites/default/files/resources/UEFI%20Firmware%20-%20Security%20Concerns%20and%20Best%20Practices.pdf)) and necessarily [protected by a strong password](https://wiki.archlinux.org/title/Unified_Extensible_Firmware_Interface/Secure_Boot#Protecting_Secure_Boot)
> > 
> > ...
> > The only way to prevent anyone with physical access from disabling Secure Boot is to protect the firmware settings with a password. Most UEFI firmwares provide such a feature, usually listed under the "Security" section in the firmware settings.

if it were simple to do, I would have done it already

jleightcap, 2023-06-21:
> if it were simple to do, I would have done it already

No doubt from me, [diff](https://github.com/tianocore/edk2/compare/master...MrChromebox:edk2:uefipayload_202304) speaks for itself concerning complexity.
Seconding the issue from a portable device security perspective.

Some resources I came across, linking here for visibility:

- https://github.com/tianocore/edk2-platforms/commit/fa878613fef54875fd1a3f300a59b60889ec651b#diff-b75dc09e50c7c8044264a59121dd5edb0c32453d3f68f4117470fca505a29394
- https://tianocore-docs.github.io/SecurityAdvisory/draft/bios_password.html

seems there's reference in-tree:

```
  MdeModulePkg/Universal/DriverSampleDxe/VfrStrings.uni
116:#string STR_PASSWORD_PROMPT            #language en-US "Set the system password"
```

MrChromebox, 2023-06-21:
sadly that's not going to work as-is for us, because it's using SMM to manage the password storage/verification, and we're not using SMM in edk2. But it's not a bad starting point

infinitelytimelesstech, 2025-12-26:
The Dasharo Team have implemented a BIOS password feature into their fork of EDK2, though I'm not sure if this would be useful here. I believe it uses SMM with a write protection option in their firmware to protect it from write access outside the firmware settings. This would be great to have here if its feasible for this project.

- https://github.com/Dasharo/edk2/blob/de7e555a0786ee4ebac0b92d867fca8f69c9696d/DasharoModulePkg/UserAuthenticationDxe/UserAuthenticationDxe.c
- https://github.com/Dasharo/edk2/blob/de7e555a0786ee4ebac0b92d867fca8f69c9696d/DasharoModulePkg/UserAuthenticationDxe/UserAuthenticationDxe.inf

nemeott, 2026-01-17:
Works great for me. Thanks for implementing a bios password!