MrChromebox/firmware #352: ASUS Chromebook Flip CX5(DELBIN) UEFI support?

Fornball, 2022-05-23:
Will there be uefi support for the ASUS Chromebook Flip CX5(DELBIN)? Currently there is only support for legacy fw.

Fornball, 2022-05-23:
I just looked at the devices list, and all the other volteer board devices do not support uefi boot yet either. Is there a specific reason for that or just low demand?

MrChromebox, 2022-05-23:
low demand. I can't imagine someone dropping $1k on one of these just to run Linux with things mostly working. That said, I have a DROBIT here and the UEFI Full Rom works fine, so if you want to test it, shoot me an email (MrChromebox at gmail) and I'll provide a link to do so

Fornball, 2022-05-23:
Alright, I will contact you in 2 weeks(I still have to use chromeos for a while).

Fornball, 2022-05-24:
> I have a DROBIT here and the UEFI Full Rom works fine  

Is there a chance this DROBIT rom will hard brick my DELBIN device? Is there a way to reflash the fw if I brick it with this rom?

a-hurst, 2022-09-29:
@Apacelus @MrChromebox Any successful reports of a UEFI ROM working properly on a DELBIN (Chromebook Flip CX5500) since this discussion?

I work in a cognitive/physio research lab, and we've been looking into picking up a bunch of cheap, large-screen, 2-in-1 laptops to study complex movement learning with a touchscreen drawing task. The Flip CX5s hit a really nice sweet spot in terms of price for its display size, specs, and build quality, but our tasks are all programmed in Python + SDL2 + OpenGL and I have no idea how well that setup will work on ChromeOS. Knowing it's possible to install a proper Linux distribution (even if I have to disable CM50 and build a UEFI image myself) would make me feel a lot better about the purchase.

Thanks in advance!

MrChromebox, 2022-09-29:
no one has tested yet, but based on my testing with DROBIT, expect that the USB-C/TB ports won't be functional

Apacelus, 2022-09-29:
@a-hurst I just ended up "enabling" rw_legacy on my cb and installing a full linux distro that way. Havent had any issues, besides usb-c ports not working and inverted screen rotation.

Sound, touch, pen, touchpad all work fine.

a-hurst, 2022-09-29:
@Apacelus Fantastic, thanks for letting me know!

When you say "inverted screen rotation", do you just mean that it won't flip the screen automatically when in tablet mode? As long as I can manually change the screen rotation that's not an issue.

Apacelus, 2022-09-30:
@a-hurst inverted means it will auto rotate... But inverted. May be a gnome issue only tho

MrChromebox, 2022-09-30:
it's a Gnome issue. you can rotate it correctly then lock it, disable it, etc

Apacelus, 2022-09-30:
* Disable via systemd -> back on after reboot
* [Set some configs](https://www.youtube.com/watch?v=Qd9_wzERFGo) -> Didnt change anything  

Unfortunately nothing except deleting/renaming the original systemd file helps.

MrChromebox, 2022-09-30:
on most Gnome distros, you can literally just select 'enable rotation lock' from the far right pull down menu after setting it correctly

Apacelus, 2022-10-01:
You can, but only after putting it in tablet mode, rotating it the other way and then locking. 
Also, it gets reunlocked after reboot/sleep.

MrChromebox, 2022-10-02:
> Also, it gets reunlocked after reboot/sleep.

I'm not seeing that problem here

eighthave, 2022-10-03:
There are cheaper "`DELBIN`" devices out there, like the Asus Chromebook Flip C536E `DELBIN-XHVI`, which is a volteer device.  Its under $500.  I had got Debian [running](https://wiki.debian.org/InstallingDebianOn/Asus/Chromebook%20Flip%20C536E) using the RW_LEGACY method, and that worked.  IMHO It can be added to https://mrchromebox.tech/#devices 

But the Chrome boot screen was still there, including the button that says to press it to reset.  One of my kids rebooted it, saw that, and it reset itself.  So now it can't boot Debian and no longer has ChromeOS.  I found this issue looking for a more permanent solution so that doesn't happen again... so I'd love to see UEFI support.

eighthave, 2022-10-04:
* _firmware-util.sh_ says `Device: ASUS Chromebook Flip CX5 (DELBIN)`. 
* The developer mode boot screen says `DELBIN-XHVI`.
* chrome says:
```
CHROMEOS_FIRMWARE_VERSION   Google_Delbin_13672.269.0
CHROMEOS_RELEASE_BOARD         volteer-signed-mp-v8keys
```
If unbrick/debug means getting a SusyQable and working with that, I can look into that.  The only source to buy them seems to be this frontshort.com but reddit threads seem to say its a scam.  I guess I need to build one then?

MrChromebox, 2022-10-04:
a DELBIN is a DELBIN. Suzy-Q cables are NLA, have to make your own

eve-horne, 2022-11-17:
> a DELBIN is a DELBIN. Suzy-Q cables are NLA, have to make your own

I've got a cheap DELBIN-XHVI and a homemade suzy-q I can test with. Been looking into building a UEFI image but if you already have a build, that would be greatly appreciated. Is there a good way to get in contact?

MrChromebox, 2022-11-17:
mrchromebox at gmail or IRC, discord

MrChromebox, 2023-01-13:
DELBIN support was added in 4.18.2

closed 2023-01-13 by MrChromebox

eighthave, 2023-02-15:
Just installed this, it worked great!  I'd just like to add one little detail: when unplugging the battery to disable the Hardware Write Protection to flash it, I pulled the cable out of the battery itself. That did not disable the Hardware Write Protection.   I had to pull the battery cable out of the motherboard to get it disabled.