MrChromebox/firmware #353: Samsung Chromebook 4 ( Gemini Lake - Bluebird ) Audio issue

CarlosTambunan, 2022-05-25:
I'm wondering why my built-in speaker is not working. Is it because of the custom firmware or the driver. for the first time I tried to install ubuntu 20.04, it worked. then when i check sound in settings, my audio is not working except via bluetooth. then i tried to change to ubuntu 18.06 but still the same. can you help me in this issue?

MrChromebox, 2022-05-25:
use a distro/release from 2022, audio will work OOTB.

closed 2022-05-25 by MrChromebox

CarlosTambunan, 2022-05-26:
Ok, thanks for your advice sir. Gonna try it when i get home, i give an update when it works

CarlosTambunan, 2022-05-26:
> I'm wondering why my built-in speaker is not working. Is it because of the custom firmware or the driver. for the first time I tried to install ubuntu 20.04, it worked. then when i check sound in [...]

CarlosTambunan, 2022-05-26:
> use a distro/release from 2022, audio will work OOTB.

Hey sir, i've installed 22.04 like you said. At the " Ubuntu Installer " windows first, the audio works. But when i update all package and driver, and then reboot the system. Suddenly the audio didn't work again. What should i do then ?

CarlosTambunan, 2022-05-26:
> I have no idea, as I don't use Ubuntu myself. For OS support, see r/chrultrabook,the Chrultrabook discord, or Ubuntu support resources. Everything is good on the firmware side of things

Ok then, thanks for your help sir

ayolinks, 2023-10-24:
Hi sir,  I  ungently need sound driver for  Samsung Chromebook 4 ( Gemini Lake - Bluebird )  Windows 10
thanks sir

MrChromebox, 2023-10-24:
> Hi sir, I ungently need sound driver for Samsung Chromebook 4 ( Gemini Lake - Bluebird ) Windows 10 thanks sir

I'm not sure why you're posting here, I have nothing to do with audio drivers, Windows or otherwise. You're looking for coolstar.org