MrChromebox/firmware #354: unsafe shutdown to ssd caused cold boot slow on kohaku

rhjdvsgsgks, 2022-05-29:
seems like nvme ssd on kohaku will not shutdown correctly. it will lose power after around 20 seconds of system shutdown rather than"normal" shutdown with system. that will cause partial data corrupted and next cold boot (boot after shutdown 20s) pretty slow (on coreboot logo, i guess it's the POST progress)

MrChromebox, 2022-05-29:
I'm not sure how this is a Chromebook firmware issue, it wouldn't cause the NVMe drive to shut down prematurely

rhjdvsgsgks, 2022-05-29:
i mean ssd shutdown too late. just like didn't received the "nvme shutdown command". ec have to force power off it after shutdown 20s

bam80, 2022-05-29:
I had sata ssd corruption after wake from suspend on Asus chromebox 3, I wonder if it's the same problem.
Do you have some evidences in system journal so I could compare?

rhjdvsgsgks, 2022-05-29:
> I had sata ssd corruption after wake from suspend on Asus chromebox 3, I wonder if it's the same problem. Do you have some evidences in system journal so I could compare?

check your smart info. after a cold boot, the "unsafe shutdown" item will increased by one

rhjdvsgsgks, 2022-05-29:
> I had sata ssd corruption after wake from suspend on Asus chromebox 3, I wonder if it's the same problem. Do you have some evidences in system journal so I could compare?

you may take a look at this first
https://bbs.archlinux.org/viewtopic.php?id=246824

bam80, 2022-05-29:
@rhjdvsgsgks I'll definitely check, thanks.
That doesn't spread to NVME I guess? I have two Asus chromeboxes 3 but the problem was registered on the one with sata ssd so far, didn't run the other long enough to check either.

rhjdvsgsgks, 2025-11-03:
seems solved  by updating to 2509.1