MrChromebox/firmware #356: CPU is throttled even if idle

bam80, 2022-06-02:
It's on Asus Chromebox 3, the CPU temp is around 40C:
[image]

MrChromebox, 2022-06-02:
what's throttling it?

bam80, 2022-06-02:
Any advice how could I debug it?

Anyway, I don't see this behavior on the same distro but different hw/fw:
[image]

bam80, 2022-06-02:
I actually couldn't provoke any throttling on a second system whatever I tried with `stress` tool.
So maybe accumulating the throttling counters is just an intended behavior when system throttles at all.

closed 2026-11-04 by MrChromebox