MrChromebox/firmware #358: PCI AER errors on Panther

hmoffatt, 2022-06-08:
I have Ubuntu with kernel 5.4 running on Asus Chromebox CN60 (Panther) and I get constant PCI AER errors.

I have installed the latest RW_LEGACY/SeaBIOS with the firmware-util script but it has not helped. Is there anything else I can do to fix this, or should I turn off PCI AER in the kernel?

```
 ChromeOS Device Firmware Utility Script [2022-05-03] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: ASUS Chromebox CN60 (PANTHER)
** Platform: Intel Haswell
**  Fw Type: Stock ChromeOS w/RW_LEGACY
**   Fw Ver:                                                                    (05/12/2014)
**    Fw WP: Disabled
*********************************************************

```

```
Jun  8 12:15:39 bandicoot kernel: [  754.201078] pcieport 0000:00:1c.0: AER: Corrected error received: 0000:00:1c.0
Jun  8 12:15:39 bandicoot kernel: [  754.201089] pcieport 0000:00:1c.0: AER: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
Jun  8 12:15:39 bandicoot kernel: [  754.201097] pcieport 0000:00:1c.0: AER:   device [8086:9c14] error status/mask=00001000/00002000
Jun  8 12:15:39 bandicoot kernel: [  754.201102] pcieport 0000:00:1c.0: AER:    [12] Timeout   
```

MrChromebox, 2022-06-08:
I'd recommend removing the WP screw and flashing the UEFI firmware. The PCI AER errors are due to the 6+ year old stock firmware you're running (RW_LEGACY is just a bootloader), and long fixed in my UEFI firmware

hmoffatt, 2022-06-08:
OK Thanks. Unfortunately I have no boot/uefi partition so I'd have to reinstall the OS too. Are the errors harmless? I think they only started with a recent kernel upgrade.

hmoffatt, 2022-06-19:
Thanks, will attempt UEFI migration some time and reinstall if necessary.

closed 2022-06-19 by hmoffatt

hmoffatt, 2022-06-20:
I migrated to UEFI (no reinstall required) and the PCI AER messages have gone. Thanks.