MrChromebox/firmware #359: CELES EC problems under Full ROM firmware?

tejasraman, 2022-06-11:
Hi @MrChromebox !
I recently installed Windows 10 on an external SD card on my Samsung Chromebook 3 (CELES). I have been getting some blue screens of death (some are related to the graphics drivers which I later fixed) but even after using chkdsk (which tried to fix my SD card), dism, sfc, and other things, I still see one message inside Event Viewer about the bluescreens:`The embedded controller (EC) did not respond within the specified timeout period. This may indicate that there is an error in the EC hardware or firmware or that the BIOS is accessing the EC incorrectly. You should check with your computer manufacturer for an upgraded BIOS. In some situations, this error may cause the computer to function incorrectly.`

Is there an EC problem with your bios under CELES? Everything else seems to work quite well under Windows 10 (it is actually pretty decent) except this, and the error message seems to suggest that this is a BIOS-related error.

tejasraman, 2022-06-12:
Turns out the BSODs were my fault - it was my grandma's Chromebook and I removed some unnecessary apps using NTLite. Reinstalling Windows stops the BSODS, the EC error still applies.

MrChromebox, 2022-06-13:
I don't see anything that would prevent the SD card reader from not powering back up when resuming from suspend - sounds more like an OS/driver issue to me

tejasraman, 2022-06-13:
Occurs under Windows 10 and Linux - seems like a firmware issue to me

tejasraman, 2022-06-13:
On Windows with coolstar's drivers, I get an error "CRITICAL PROCESS DIED" even on a clean Windows To Go install with Rufus I try.

On Linux, I see many IO errors that suggest /dev/mmcblk1 is not there (again, that's the sad card)

Brunch on SD card storage has the same problems as Linux

Brunch on eMMC works fine

MrChromebox, 2022-06-13:
I have no idea what drivers exist/are being used for the SD card reader; coolstar certainly hasn't written any.

what UEFI firmware version are you using?

tejasraman, 2022-06-13:
**your Full ROM UEFI firmware**
This occurs under linux/brunch too
I am on version 4.16 of coreboot if you want to know

MrChromebox, 2022-06-13:
> I am on version 4.16 of coreboot if you want to know

that's what I was asking

tejasraman, 2022-06-15:
Ok... ***but why?***

MrChromebox, 2022-06-15:
fuck if I know. CELES has been an absolute shitshow, regardless of firmware type used. I recommend users leave it running ChromeOS for a reason.

tejasraman, 2022-06-15:
The OSes I tried worked quite well but the sad card reader was a problem on every one. 

I gave up on the sad card and out brunch on the eMMC - had some audio issues with bad UCM config files, submitted a PR to the brunch dev - it got accepted.

I'll use the SD card for something else, but I wouldn't have gotten it if I had known that the SD card slot would mess with me

CELES worked out of the box on every Ubuntu 22.04 Linux I tried, including audio, but not suspend (because of that SD card lol)

closed 2022-06-15 by tejasraman

tejasraman, 2022-07-23:
This was my grandmother's chromebook (she only uses web browsing, YT and Duolingo) and I stopped using the chromebook and bought a surface laptop for for her (on sale for $299 + discounts).

Suffice to say, I'm done with the chromebook.