MrChromebox/firmware #360: Asus Chromebox 3 HDMI no audio output

JustATester2, 2022-06-18:
Device: ASUS Chromebox 3 (CN65) both 3865U and 8550U version tested
Fw Ver: MrChromebox-4.16 (27/04/2022) 
OS: WIN10/WIN11
Well done for windows usb issue fix, now the box is working pretty well.
However, there is no audio output via HDMI ,audio output via USB-C/Headphone works well.
Driver version
Intel SST 9.21.0.3869
Realtek 5663 HW Audio Codec 10.0.10586.6834
Intel HD Graphics 610/620 30.0.101.1340
Intel (R) Display-Audio 10.27.0.11
It's weried that intel graphics command center shows the display connector-type is Displayport when HDMI connected.

Also tested on Ubuntu 20.04 LTS usb live boot
The arandr command shows the display was connected via HDMI-1, but there is only dummy output, HDMI audio not show either.

USB-C has audio output while HDMI dosen't.
Any ideas how to enable HDMI audio output?

Thanks.

MrChromebox, 2022-06-18:
nope, just use USB-C -> HDMI

MrChromebox, 2022-06-18:
FWIW, HDMI audio works under LibreELEC (Linux) so it's an OS/driver issue, not a firmware one

MrChromebox, 2022-06-18:
looks like you can force the snd_hda_intel driver to attach for HDMI via:
`sudo nano /etc/modprobe.d/audio.conf` (will create file)
add line:
`options snd-intel-dspcfg dsp_driver=1`
save (CTRL+o, CTRL+x) and reboot

JustATester2, 2022-06-21:
Thank you for your reply.
Forget about Windows.
LiberELEC is good enough for me.
I installed Hyperion.ng on the box and WLED light work perfectly!

closed 2022-08-24 by MrChromebox

ACR-Jeff, 2023-03-08:
I know this post is old, Yet I had the same issues and found a solution on my end, My Audio in Debian 11 and Debian 12 was null and No Audio Devices found. I figured it may help someone else in the future

Download and Install Firmware-sof 1.7-1
[http://http.us.debian.org/debian/pool/non-free/f/firmware-sof/firmware-sof-signed_1.7-1_all.deb](url)

After installing open terminal and execute this command:
`echo "options snd-hda-intel dmic_detect=0" | sudo tee -a /etc/modprobe.d/alsa.conf`

Restart your computer

Notice: Older Hardware is not supported any further after 2.x versions of Firmware-sof
At [https://salsa.debian.org/mpearson/firmware-sof ](url)

wruberte, 2023-05-05:
### Asus Chromebox 3 (N65) Chrome OS Flex No HDMI Audio Fix

Thanks @ACR-Jeff  and @MrChromebox for the previous guidance to put the pieces together in a consolidated read-me.

#### Pre-requisites
 1. Remove Read-Only Firmware Screw 
 2. [MrChromebox Firmware installation](https://mrchromebox.tech/#fwscript) using _Install\Update UEFI (Full ROM) Firmware_ option
 3. Enable Developer Mode in ChromeOS Flex
     1. [Read Enable Developer Mode in ChromeOS Flex](https://shibumi.dev/posts/enable-developer-mode-on-chrome-os-flex/)
     2. Re-start and boot into a live distribution of Linux on USB
     3. _Presumption:_ Internal Drive has Chrome OS Flex installation and is mapped to /dev/sda* 
         - `sudo mount /dev/sda12 /mnt`
     4. Locate _grub.cfg_ and edit replacing _cros_efi_ with _cros_efi cros_debug_
     5. save and reboot

#### Enable Read-Write File System
  1. Open Chrome Browser and Launch Shell by pressing Ctrl+Alt+T
  2. `shell`
  6.  `sudo /usr/share/vboot/bin/make_dev_ssd.sh --remove_rootfs_verification --force --partition 2`
  7. Restart
  8. Open Chrome Browser and Launch Shell by pressing Ctrl+Alt+T
  9. `shell`
  10. `sudo mount -o remount,rw /`

#### Enable Developer Mode in ChromeOS Flex
 1. `echo "options snd-hda-intel dmic_detect=0" | sudo tee -a /etc/modprobe.d/alsa.conf`
 2. restart

#### Validation
 1. Launch Settings=>Device=>Audio
 2. In speaker drop down select form field, validate that _HDMI/DP_ appears

#### Warning

In the spirit of Dr Strange, _You really should stole the whole book because the warnings come after the spells_, the following comes at the end of this read-me:
- The above worked as of May 4, 2023
- Your mileage may vary
- This definitely voids the warranty
- Given this model of Chromebox launched in 2018 you will be lucky to still have a warranty
- I am not responsible for any damage performed on your system

MrChromebox, 2023-05-05:
@wruberte why are you running Flex on the CN65? It's still supported for official ChromeOS

wruberte, 2023-05-06:
> @wruberte why are you running Flex on the CN65? It's still supported for official ChromeOS

@MrChromebox, thank you for following up.  Original intent was to install Fedora 38 to an external USB Drive while preserving the original ChromeOS.  Detail provided below regarding failed boot of ChromeOS in Dev Mode from internal SSD using UEFI Full ROM Firmware.

1. **mrchrombox UEFI Install** 
    - Installed the latest firmware using the _Install\Update UEFI (Full ROM) Firmware_ option following steps from [Computer Headquarters Guide for Installing Windows 10 on a Asus Chromebox 3 Guide](https://www.youtube.com/watch?v=azxu6-_b5Bs) leading up to 6:41m mark on video
2. **ChromeOS Boot Failure after firmware install** 
    - After firmware install, tested booting from the internal SSD that had ChromeOS unsuccessfully encountering an error similar to what was recorded on the 7:52 mark on the video
    - **Error from video:** `Booting from SATA:TS32GMTS400  failed.`
    - **Error encountered:** `Booting from 'SATA: TS64GMTS400' failed: verify it contains a 64-bit UEFI OS.`
    - Proceeded to install Fedora 38 to external USB with the expected results
3. **Failed to boot ChromeOS Recovery USB**
    -  Circled back and created a ChromeOS Recovery USB matching Asus Chromebox 3 using [Chromebook Recovery Utility](https://chrome.google.com/webstore/detail/chromebook-recovery-utili/pocpnlppkickgojjlmhdmidojbmbodfm)
    -  Boot of ChromeOS Recovery USB failed with blank screen
4. **ChromeOS Flex Install**
    - Created a ChromeOS Recovery USB matching the ChromeOS Flex using the [Chromebook Recovery Utility](https://chrome.google.com/webstore/detail/chromebook-recovery-utili/pocpnlppkickgojjlmhdmidojbmbodfm)
    - Boot of ChromeOS Recovery USB matching the ChromeOS Flex worked permitting installation to internal SSD

MrChromebox, 2023-05-06:
@wruberte my recollection is a bit hazy, but I believe the CN65 has functional legacy boot mode, which is designed to do exactly what you want - ChromeOS + Linux. However, if you have Flex running well then no harm in sticking with it

dadr, 2023-06-15:
I want to thank @ACR-Jeff and leave a hint about what worked for me.  I have HP Chromebox 3 (NOIBAT) with coreboot 4.20 and am running Ubuntu 23.04 - using secure boot!   I had no sound, and this added it to the HDMI output, which was what I wanted.  I did not install any firmware, just added the following line to `/etc/modprobe.d/alsa-base.conf`
`options snd-hda-intel dmic_detect=0`

The only other quirk I have is that the system locks up when the screen is blanked.   I can suspend and resume just fine, but if I set a screen blank interval and it's reached, then the screen gently gets darker, and the whole machine locks up, needing a power cycle to recover.

Lastly, since it's the first time I've commented here, I want to give a big thank you to @MrChromebox for working on and sharing the coreboot firmware.

e-Surfer, 2023-12-28:
> Device: ASUS Chromebox 3 (CN65) both 3865U and 8550U version tested Fw Ver: MrChromebox-4.16 (27/04/2022) OS: WIN10/WIN11 Well done for windows usb issue fix, now the box is working pretty well. [...]

Hi togehter,

I´ve also switched my ASUS Chromebox 3 to Windows 10.
Problem is that die Audio device is not detected correctly and no manuel installed driver works.

Can anybody support here? Thx a lot in advance.

Kind Regards
Christian

[image]

MrChromebox, 2023-12-28:
@e-Surfer for Windows support, please use the Chrultrabook forums: https://forum.chrultrabook.com/

my github is for firmware issues only, not OS/drivers