MrChromebox/firmware #361: Panther won't boot if HDMI device switched off

hmoffatt, 2022-06-19:
I have my Chromebox (Panther) connected to my TV. It won't boot if the TV is not turned on - it just hangs. I'm using Seabios / legacy firmware. Is this something that might be fixed with the full UEFI firmware?

MrChromebox, 2022-06-19:
the UEFI firmware is unequivocally better, but I can't guarantee it will boot without a display connected (which is essentially what you are trying to do).

hmoffatt, 2022-06-19:
It worked perfectly with my old TV, just not the new one. Even switching the TV after power up doesn't help.

MrChromebox, 2022-06-19:
because your TV isn't sending a valid EDID to the box, most likely. There's not much you can do short of buying a box that caches the EDID and provides it to the Chromebox when the TV is off. Or modifying your behavior to only power on the box when the TV is already on

hmoffatt, 2022-06-20:
I've just installed the UEFI firmware (to fix my PCI AER errors) and now the Panther does boot with the TV switched off. Brlliant, thanks.

closed 2022-06-20 by hmoffatt