MrChromebox/firmware #363: Windows ACPI-BIOS-ERROR UEFI

INeedHelpppp, 2022-07-02:
Device: ASUS Chromebook FLIP c436fa
Whenever booting in windows it would blue screen ACPI-BIOS-ERROR
Is there anybody can help with this problem?

sky-ranch, 2022-07-05:
Hi, i have a same problem

i run script at My Galaxy Chrome book (KoHAKU, comet lake) and sucessfully install UEFI

but when i trying to install Windows 10 x64
it shows only ACPI_BIOS_Error with Blue Screen

i Also tried install x64 Windows 11, but it show ACPI_BIOS_Error again

(cf, the UEFI was 220704 version)

MaCleodWalker, 2022-07-05:
I'm currently having the same issue with my Galaxy Chromebook. According to olm3ca's guide 10 and 11 can be installed but I cant get past the ACPI_BIOS_Error

closed 2022-07-15 by INeedHelpppp