MrChromebox/firmware #364: Chromebook Pixel 2013 (link) headphone jack issue

elektroencefalografista, 2022-07-03:
After connecting headphones through a 3.5mm headphone jack they're being correctly detected, the system volume level changes, but the audio is still being routed to internal speakers. The headphone output stays quiet. The issue persists on Cloudready, brunch, and Linux distros. I'm on the newest firmware.
I tried using the chromebook_audio=1 option, but this only makes all audio devices disappear from the volume dropdown. I also tried using different kernels (mainstream and chromebook specific) to no avail.

MrChromebox, 2022-07-03:
nothing has changed on LINK w/r/t audio in years; this sounds like a kernel/driver issue, rather than a firmware/HBA verb config one

renr4, 2022-08-24:
Check that alsamixer -> HP/Speaker Auto Detect is On

gsstark, 2022-12-13:
Fwiw I see the same behaviour. This is on a 2013 Pixel that is freshly flashed with firmware and freshly installed Chromeos Flex (aka Cloudready). Plugging in a wired headphone causes the GUI to pop up the volume level widget so it's detecting the headphones but the audio continues to actually be routed through the speakers. 

The speakers also don't respond to the volume level, they are always on full volume or mute regardless of the setting in the GUI. 

Unfortunately I don't have a linux boot available so I'm limited to the tools in Chromeos, if there's any way I can debug this.

coolstar, 2023-06-30:
Headphone issue is an issue with kernel not sending EAPD HDA verb (to node 0x10) after loading codec firmware

MrChromebox, 2023-07-21:
This was possibly fixed in the 4.20.1 release, please retest with it

coolstar, 2023-07-23:
Flashed + booted Windows 10 20H2 here on Link w/ Creative Recon3Di driver. Can confirm speakers and headphone jack both functional

closed 2023-07-23 by MrChromebox

renr4, 2023-09-06:
> Headphone issue is an issue with kernel not sending EAPD HDA verb (to node 0x10) after loading codec firmware

Is there a way I can send that verb manually somehow? Just installed Cloudready 96 and everything works perfectly except that headphone jack issue.

drafnel, 2026-02-24:
Update: it's working after using alsamixer to enable "HP/Speaker Auto Detect".

Here are the steps:
* Run alsamixer
* Hit F6 to select sound card
* Select "HDA Intel PCH"
* Use right arrow to move to "HP/Speaker Auto Detect"
* Hit 'm' to "unmute" it i.e. enable it
* Hit Escape to exit.
* Headphone jack should work now.
* Settings are persisted after reboot.