MrChromebox/firmware #366: UEFI support for Chromebase 22" All In One

SuperKnux, 2022-07-16:
Model Board: DOOLY
Base Board: Puff
Processor: Intel Pentium 6405U (Comet Lake)

I dunno what else to put in here, so if there's anything I'm missing, tell me.

MrChromebox, 2022-07-16:
just added RW_LEGACY support, LMK if that works. Don't want to add an untested UEFI Full ROM firmware for it unless you have a Suzy-Q cable to unbrick

closed 2022-08-24 by MrChromebox

nangiloc, 2023-06-16:
I own this chromebase, found the jumper on the motherboard.

RW_LEGACY works, I installed Arch instead of Chrome OS.

I don't know if I missed something, but the live (installation media) boots in UEFI mode. I installed Arch Linux by formatting the SSD to GPT (with a bios boot partition). I use GRUB. I tried both GRUB installation methods, BIOS/GPT and UEFI.
The first did nothing, the second worked. But the bootloader does not launch GRUB automatically. If I let it, it shows an error that there is no UEFI OS on the SSD. I have to select "Boot from file", navigate to /boot/grub/x86_64-efi/grub.efi to launch Arch.

This means I have to keep a USB keyboard plugged in every time I want to boot the computer.

I'd love to test a UEFI Full ROM but I don't own a Suzy-Q cable.

nangiloc, 2023-06-16:
Found a solution here : #409

letrain02, 2024-04-12:
> I own this chromebase, found the jumper on the motherboard.

where is the jumper? i've googled and googled... and i'm probably just not typing in the right words.

MrChromebox, 2024-04-12:
all of the 10th-gen/puff-based Chromeboxes use a similar board layout. My supported devices page shows the board / jumper location for several of them -- use those as a reference