MrChromebox/firmware #367: FN functions keys not working on acer C733-C607 (bobba)

malbarbo, 2022-07-20:
I installed the full UEFI firmware (version 4.16) and it is working fine on my chromebook, but I have a issue with the keyboard.

If I configure X11 to use the "chromebook" keyboard layout, the top row generated the expected keys KEY_BACK..KEY_VOLUMEUP, but with I choose pc layout, the same keys are generated, but the expected are F1..F10.

I've tested my setup with an external keyboard and it work as expected. The F1 key on the external keyboard generated F1 with pc layout and generate KEY_BACK with chromebook layout. So, I think that it is a firmware issue.

MrChromebox, 2022-07-20:
> I choose pc layout, the same keys are generated, but the expected are F1..F10

why are you expecting F1..F10? There is no label on the keys as such. The fact that they occupy the same location as F-keys would on a PC keyboard does not make them so.

Older Chromebooks did generate the scancodes for F-keys and ChromeOS (or Linux) mapped them to the correct functions. Newer Chromebooks generate unique scancodes for the top row keys, which the kernel HID driver maps accordingly.

I could disable this in firmware, but there would be side effect either way. I'll look into a way to have it toggleable.

malbarbo, 2022-07-20:
> why are you expecting F1..F10? There is no label on the keys as such. The fact that they occupy the same location as F-keys would on a PC keyboard does not make them so.

Well, I have an older Chromebook and it generate these keys, that's why I was expecting it. I didn't knew there was difference between older and newer Chromebooks as you said.
 
> I could disable this in firmware, but there would be side effect either way. I'll look into a way to have it toggleable.

That you be great! By the way, thanks for your amazing work.

dewitte-77, 2023-01-02:
You can re-assign the scancodes by using udev rules.
Then you can also generate the necessary shortcuts using the "keyd" daemon.
[https://wiki.archlinux.org/title/Lenovo_IdeaPad_Flex_3_CB_11IGL05_Chromebook#Function_keys](url)

closed 2026-11-04 by MrChromebox