MrChromebox/firmware #368: Black screen after flashing full firmware Samsung Chromebox 3 XE300M22

zmeishe1984, 2022-07-21:
Good afternoon, after flashing using a script, Chromebox does not load
Black screen after power on
I flashed both with a script and ch341a
The write protection jumper is disabled, the firmware was successful in all cases
flashed with this firmware: coreboot_tiano-stumpy-mrchromebox_20220718.rom

MrChromebox, 2022-07-21:
works perfectly well here on my STUMPY. What display connection are you using? RAM config?

zmeishe1984, 2022-07-22:
On the stock bios everything works fine. Memory inserted into different slots.
Display connected via DVI (via adapter)

zmeishe1984, 2022-07-22:
> works perfectly well here on my STUMPY. What display connection are you using? RAM config?

What firmware version did you flash?

if it differs from the current one, please put it somewhere on a file sharing service

MrChromebox, 2022-07-22:
the public 4.17.2 release (7/18/22) via the public script. Using DVI->HDMI.

zmeishe1984, 2022-07-24:
Problem solved
installed bios from here: johnlewis.ie/Chromebook-ROMs/stumpy/coreboot-stumpy-seabios-310714.rom
Everywhere they write that seabios is not supported on my device, but everything worked fine...

MrChromebox, 2022-07-24:
> installed bios from here: johnlewis.ie/Chromebook-ROMs/stumpy/coreboot-stumpy-seabios-310714.rom

IMO, running ancient, unsupported firmware from 2014 is not the solution.

> Everywhere they write that seabios is not supported on my device, but everything worked fine...

no, everywhere they write that legacy boot mode is not supported, which is correct. **I** simply don't build/distribute legacy BIOS full ROM firmware

I can build you a debug version of my current UEFI firmware, which you can flash, boot, read, and extract a debug log from, should you care to.

zmeishe1984, 2022-07-27:
I wonder why it works for some, but it didn't work for me? maybe there is some difference in the devices?

It would be nice to make a debug version, maybe soon you will need to flash the same device again...

closed 2023-01-13 by MrChromebox