MrChromebox/firmware #370: Pixel Slate ACPI BIOS ERROR

crftbt, 2022-07-22:
Attempting install of Windows 10 (and other OS options), receiving an ACPI BIOS ERROR on a Google Pixel Slate. 

Tried both [coreboot_tiano-nocturne-mrchromebox_20220708.rom](https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-nocturne-mrchromebox_20220708.rom) and https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-nocturne-mrchromebox_20220718.rom. 

What can be done to address the ACPI BIOS ERROR?

MrChromebox, 2022-07-22:
well, someone can figure out the regression. Or at the very least, boot Linux and use `sudo acpidump -b` and provide the dsdt and ssdt files

crftbt, 2022-07-22:
https://www.dropbox.com/s/228v75jagbu6lw2/ssdt.dat?dl=1
https://www.dropbox.com/s/pow2whey7dizw3w/dsdt.dat?dl=1

crftbt, 2022-07-23:
From fwts results.log:

Test Failure Summary
================================================================================

Critical failures: 2
 mtrr: Memory range 0x27f000000 to 0x27f000fff (0000:00:1e.0) has incorrect attribute Write-Back.
 mtrr: Memory range 0x27f001000 to 0x27f001fff (0000:00:1e.0) has incorrect attribute Write-Back.

High failures: 6
 klog: HIGH Kernel message: [    0.113192] ACPI Error: AE_NOT_FOUND, While resolving a named reference package element - FCPR (20210730/dspkginit-438)
 klog: HIGH Kernel message: [    0.113247] ACPI Error: AE_NOT_FOUND, While resolving a named reference package element - RCPR (20210730/dspkginit-438)
 dbg2: DBG2 Generic Address Structure has zero register bit width which is probably incorrect
 uniqueid: \_SB_.PCI0.SPI0.S001._HID/_UID conflict with \_SB_.PCI0.I2C3.NVM0._HID/_UID
 uniqueid: \_SB_.PCI0.SPI1.CRFP._HID/_UID conflict with \_SB_.PCI0.I2C5.NVM1._HID/_UID
 dmicheck: Type 17 expects length of 0x5c, has incorrect length of 0x28

Medium failures: 5
 autobrightness: 1 brightness levels did not match the brightness level just set for backlight intel_backlight.
 fadt: FADT P_LVL2_LAT is <= 100 (0) which implies a C2 state is supported but there is no P_BLK register block defined to enable the C2 transition.
 fadt: FADT P_LVL3_LAT is <= 1000 (0) which implies a C3 state is supported but there is no P_BLK register block defined to enable the C3 transition.
 method: \_SB_.PCI0.I2C4.MAXR._HID returned a string 'MX98373' but it was not a valid PNP ID or a valid ACPI ID.
 method: \_SB_.PCI0.I2C4.MAXL._HID returned a string 'MX98373' but it was not a valid PNP ID or a valid ACPI ID.

Low failures: 5
 mcfg: MCFG MMIO config space at 0xe0000000 is not reserved in the memory map table
 dmicheck: String index 0x02 in table entry 'Memory Device (Type 17)' @ 0x7aa411d4, field 'Serial Number', offset 0x18 has a default value '00000000' and probably has not been updated by the BIOS vendor.
 dmicheck: String index 0x02 in table entry 'Memory Device (Type 17)' @ 0x7aa41252, field 'Serial Number', offset 0x18 has a default value '00000000' and probably has not been updated by the BIOS vendor.
 dmicheck: String index 0x02 in table entry 'Memory Device (Type 17)' @ 0x7aa3f1d4, field 'Serial Number', offset 0x18 has a default value '00000000' and probably has not been updated by the BIOS vendor.
 dmicheck: String index 0x02 in table entry 'Memory Device (Type 17)' @ 0x7aa3f252, field 'Serial Number', offset 0x18 has a default value '00000000' and probably has not been updated by the BIOS vendor.

Other failures: NONE

MrChromebox, 2022-07-24:
sadly fwts has no correlation with WHQL firmware requirements.

that's said, a HID/UID conflict would definitely make Windows grumpy, so I'll take a look at that

MrChromebox, 2022-07-24:
ok, shoot me an email (MrChromebox at gmail) and I'll whip up a test build for you to try and see if that resolves the problem

crftbt, 2022-07-29:
Are rom files older than 20220708 archived anywhere online?

MrChromebox, 2022-07-29:
nope

MrChromebox, 2022-08-05:
fixed with 08/05/2022 release

closed 2022-08-05 by MrChromebox