MrChromebox/firmware #371: Lenovo X131e "stout" BIOS ACPI ERROR

agnivo007, 2022-07-25:
Updated to latest 4.17.2 for stout. Existing Win10 install doesn't boot and keep restarting with stop code BIOS_ACPI_ERROR.

Tried installing 7/8/10 via USB flash, same stop code and BSOD.

Can boot to MX Linux and ChromeOS Flex.

Please advise.

MrChromebox, 2022-07-25:
boot Linux, install acpica-tools pkg, run `sudo acpi-dump -b` and attach the dsdt and ssdt files. Also run Ubuntu's firmware test-suite (fwts) and post results.

agnivo007, 2022-07-26:
[acpi.zip](https://github.com/MrChromebox/firmware/files/9186055/acpi.zip)
Uploaded acpidump, don't have an ubuntu iso right now. Will the above work, or fwts is mandatory?

scaniazero, 2022-07-26:
having the exact same problem with my Lenovo X131e (Stout) after updated the firmware with Ubuntu to 4.17.2 i get the same exact message ACPI_BIOS_ERROR loop when i tried to install Windows 10 x64 21H2 with USB Bootable.
[fwts-stout.txt](https://github.com/MrChromebox/firmware/files/9189455/fwts-stout.txt)
uploaded fwts from my X131e

agnivo007, 2022-07-26:
@scaniazero Did you run it using sudo ?

agnivo007, 2022-07-26:
I hope all required files are uploaded by @scaniazero and myself for you to inspect.

agnivo007, 2022-07-29:
@MrChromebox Please give a reply if you have seen the uploads. I'm not pushing you in any way, but this error has completely made windows inoperable which my child needs for school stuff. Thanks.

MrChromebox, 2022-07-29:
found the culprit in coreboot, trying to find a proper solution vs reverting

agnivo007, 2022-07-31:
@MrChromebox Ok, awaiting a fix and further instructions eagerly. Thanks.

agnivo007, 2022-08-07:
@MrChromebox Also, would like to inform that updates for about 1.5yrs or so have probably made the system loose control over fan speed whatever the OS be. It just always spins at the lowest speed on the latest firmware and the lower body plate heats up like a frying pan (Linux/ChromeOS flex).

This model is already 10 years old and if you want to drop further development on this, please let us revert to an earlier build that atleast worked with windows 10 and offered proper acpi and fan control on any OS.

Thanks and hoping for a fix soon...

MrChromebox, 2022-08-07:
I don't keep earlier builds, but I also don't believe anything related to the fan control has changed (I've looked into this before).

I pushed an update to fix the Windows ACPI error the other day.

agnivo007, 2022-08-08:
Ok, was able to install win 10 after updating. Will see how it goes... Thanks.

closed 2022-08-24 by MrChromebox