MrChromebox/firmware #373: Lenovo x131e Stout

Deguello-7, 2022-08-11:
Hi;
I disabled WP, installed full ROM firmware. I removed the 16G SSD and installed a 128G. The machine boots to shell, when I attempt to boot from the Win 10 ISO, it starts to load Windows  then I get the sad face blue screen and reboots. Did I need to prep the new SSD? Help please!

MrChromebox, 2022-08-11:
boot to UEFI settings screen. What firmware version/date is listed? If not the latest (4.17.2 - 8/04/2022) then you need to boot a Linux USB and update it

Deguello-7, 2022-08-11:
FW is 4.17.2-07/18/2022

MrChromebox, 2022-08-12:
please boot a Linux USB and update the firmware using my script, that will fix the BSOD

Deguello-7, 2022-08-12:
I am downloading Ubuntu 22.04.1 amd 64. Will that work?

MrChromebox, 2022-08-12:
it should

Deguello-7, 2022-08-18:
I had time to get back to this tonight. I ended up getting Linux Lite and it is working . I am installing Win 10 right now. Thanks for your help!!

closed 2022-08-19 by MrChromebox