MrChromebox/firmware #375: Lenovo LICK full UEFI not detecting internal storage

walmartshopper, 2022-08-16:
I just got a Lenovo chromebook 3 (boardname LICK) and flashed the full UEFI.  I booted the Arch Linux installer USB and was able to install the system to the /dev/mmcblk0 drive and create an efi boot entry.  However when I try to reboot into the installed system, it fails. If I go into the UEFI settings, I notice the internal mmc drive does not show up anywhere. If I select Add Boot Option or Boot From File, it's just blank.

HP Chromebook x360 11 G2 EE (Lick)
FW: MrChromebox-4.17.2 07/18/2022

walmartshopper, 2022-08-16:
As a workaround, I installed grub onto an old 4GB microsd card.  I can boot grub from the microsd card, and then grub can boot my Arch install on the internal emmc drive.  It's not ideal, but I wasn't really planning on using the microsd slot so this will work for now.

MrChromebox, 2022-08-16:
try this: boot Linux, run script, clear NVRAM, reboot

there's no reason the internal eMMC shouldn't be working on LICK

walmartshopper, 2022-08-17:
cleared NVRAM from inside arch, internal emmc still not showing up.

MrChromebox, 2022-08-18:
can you pull a cbmem log?

walmartshopper, 2022-08-19:
```
[root@chromebook-small Downloads]# ./cbmem 
31 entries total:

   0:1st timestamp                                     156,750
  11:start of bootblock                                160,111 (3,360)
  12:end of bootblock                                  163,390 (3,279)
  13:starting to load romstage                         163,390 (0)
  14:finished loading romstage                         169,763 (6,373)
   1:start of romstage                                 169,770 (6)
 970:loading FSP-M                                     169,938 (167)
   2:before RAM initialization                         211,938 (42,000)
 950:calling FspMemoryInit                             218,687 (6,748)
 951:returning from FspMemoryInit                      274,942 (56,254)
 550:starting to load Chrome OS VPD                    275,365 (422)
 551:finished loading Chrome OS VPD (RO)               275,378 (12)
   3:after RAM initialization                          275,439 (61)
   4:end of romstage                                   277,725 (2,286)
 100:start of postcar                                  278,738 (1,012)
 101:end of postcar                                    278,738 (0)
  10:start of ramstage                                 278,846 (107)
  30:device enumeration                                292,588 (13,742)
 971:loading FSP-S                                     304,039 (11,451)
 954:calling FspSiliconInit                            305,010 (970)
 955:returning from FspSiliconInit                     342,168 (37,158)
  40:device configuration                              349,686 (7,518)
 956:calling FspNotify(AfterPciEnumeration)            354,334 (4,647)
 957:returning from FspNotify(AfterPciEnumeration)     354,584 (250)
  50:device enable                                     354,590 (5)
  60:device initialization                             594,462 (239,871)
  70:device setup done                                 599,181 (4,719)
 958:calling FspNotify(ReadyToBoot)                    599,628 (446)
 959:returning from FspNotify(ReadyToBoot)             599,749 (120)
 960:calling FspNotify(EndOfFirmware)                  599,749 (0)
 961:returning from FspNotify(EndOfFirmware)           604,231 (4,481)
  98:ACPI wake jump                                    609,256 (5,025)

Total Time: 609,240
```

MrChromebox, 2022-08-19:
no issues with coreboot init, must be a Tianocore issue

walmartshopper, 2022-08-20:
Thanks for looking into this. These are selling for $79 right now on best buy and were recently posted on a deal website, so I imagine there might be some more users in the near future.  I had enough success with it that I ordered two more of them for kids school stuff.  The only only issue I have had besides this boot issue is the headphone jack not working.  Internal speakers and microphone work when using sof-firmware with pipewire and the pro audio profile. Webcam, touchpad, brightness controls, suspend all work fine.  And I can use a cheap USB audio dongle if headphones are needed.

MrChromebox, 2022-08-20:
well, if you can reproduce on all 3 units, perhaps you can lend me one for troubleshooting. The two GLK devices I have here have no eMMC issues, and I haven't heard of anyone else having any either

walmartshopper, 2022-08-24:
I could not reproduce on the 2 new units, the eMMC showed up fine.  On the non-working unit, I tried re-flashing and clearing nvram again, still doesn't show up. The only thing different between them is that the first one had been fully updated to the latest Chrome OS before I flashed it.  The two that are working were flashed right out of the box without updating Chrome OS first.

closed 2022-08-24 by walmartshopper