MrChromebox/firmware #377: Acer 317 (Jasper) GRUB has no "search"?

wesmorrison, 2022-08-25:
So I've looked at the scripts and in /tmp and I believe this installed:

rwl_altfw_jsl-mrchromebox_20211115.bin

When I boot and choose alternate, TianoCore, then c for GRUB shell, then TAB for commands, there is no "search"

Is there an available firmware that will work on Jasper that does have search? Here's why.

My drives change if I have a USB stick plugged in. If none, the root I want is on hd0. If I have a stick in, that stick is hd0 and the root I want is on hd1. So if I set up menu entries they won't work depending on what's plugged in.

I want to use "search" to find the correct drive rather than set root=(something)

I installed Lubuntu with chrx (it repartitioned main drive) but it didn't set up a way to boot it easily. This Chromebook has different boot menus than I see described elsewhere, no CTRL-D or CTRL-L. I have to enter GRUB and set root/linux/initrd/boot every time.

Willing to try alternates if I knew where to get them. RW_LEGACY only affects "not chromebook" booting, right? If RW_LEGACY gets messed up, I can still boot ChromeOS? I also have a stick with "Breath" that boots when I choose "boot from external drive" so hopefully I'll have a way to reflash if necessary.

MrChromebox, 2022-08-25:
grub is not a part of Tianocore. Either you're referring to the UEFI shell (which is pretty obvious, since it says 'welcome to the UEFI shell'), or you're booting grub off the internal drive (an unused remnant of ChromiumOS).

chrx can only be used to install an OS in legacy boot mode, not UEFI boot mode, which is what newer Chromebooks like yours use for "legacy" boot mode. You need boot your OS installer from USB via Tianocore, then reinstall your OS from USB to the partition resized by chrx (p7) and ensure the UEFI bootloader is installed to p12.

wesmorrison, 2022-08-26:
Thank you, that gives me some ideas to try.

> grub is not a part of Tianocore.

(Smacks forehead) You are correct. I forget all this stuff when I don't do it every day. Programmer by trade, build/modify computers occasionally.

> You need boot your OS installer from USB via Tianocore, then reinstall your OS from USB to the partition resized by chrx (p7) and ensure the UEFI bootloader is installed to p12.

Are you saying Chromebook doesn't need p12 to boot normally into Chromeos? I can let OS installer put its bootloader there?

Sorry this didn't turn out to actually be a firmware issue.

MrChromebox, 2022-08-26:
> Are you saying Chromebook doesn't need p12 to boot normally into Chromeos? I can let OS installer put its bootloader there?

correct, ChromeOS does not use p12 at all.

MrChromebox, 2022-09-05:
closing as not a firmware issue

closed 2022-09-05 by MrChromebox