MrChromebox/firmware #380: Samus Chromebook Pixel 2015, No Charging on Latest Firmware

joshaspinall, 2022-09-03:
Hello,

I have just used your tooling, documentation and scripts to revive an older Chromebook Pixel.  It runs fantastically under a couple of Linux flavours I have tried.

Unfortunately, the machine will not charge or draw power from the wall under any condition using the latest build.  The machine was fully charged prior to flashing the firmware and installing UEFI Linux.  The firmware was recovered to the backup taken during flashing, and after rebooting the machine immediately began to charge.  Flashing again via the script utility, immediately on reboot charging is not possible.

I am unsure if this is an issue with my machine or a bug with the firmware, as I do not have another device available to test the current release.

Please contact me if I can assist further, thanks!

MrChromebox, 2022-09-04:
please check the cbmem log after booting and see which EC firmware is active (1 is RO, 2 is RW). If RO is active that would explain the issue. A hard reset (refresh + power) may fix an EC that is stuck in RO

joshaspinall, 2022-09-10:
Thanks for the response.

Having studied both yours and the Coreboot documentation, I am unsure how to check the logs as you describe.  If you are able to provide some guidance, I am more than happy to follow and report the results.  If the guidance already exists, I have been unable to find it!

Since initially filing the report, charging has been intermittent on the device.  Does a refresh + power have the same effect when using full UEFI Coreboot?  Again, couldn't find anything in the documentation.  Happy to raise a pull request if it helps out!

MrChromebox, 2022-09-10:
download and extract: https://mrchromebox.tech/files/util/cbmem.tar.gz  
run: `sudo ./cbmem -1 > cbmem.log`
throw log on pastebin and post link here

a hard reset is a function of the EC and works the same with stock or my upstream coreboot firmware

joshaspinall, 2022-09-11:
Thanks for taking the time to provide guidance.  

Output of the log linked [here](https://pastebin.com/4iv0eFHX).

MrChromebox, 2022-09-11:
everything looks good there. I don't have my SAMUS at the moment, I loaned it to coolstar, but they didn't report any such issues on the latest firmware so IDK

joshaspinall, 2022-09-14:
Interesting.  Many thanks for taking the time to look.  

In summary, the behavior is different between the stock firmware and the updated, varying also from charger to charger.  However, all of the chargers I have tried work as expected on the stock firmware.  

Please let me know if there is anything I can do to investigate further, though it doesn't sound like a widespread problem.

navyfighter12, 2022-09-18:
I'm running the latest firmware on my Samus Chromebook 2015 and no issues charging. I tested 2 chargers and all working perfectly. I did install a few distros and kernels but I am unable to replicate the no charging issue. If you need to compare anything between ours let me know.

closed 2023-01-13 by MrChromebox