MrChromebox/firmware #382: Alder-Lake support (?)

maxiauer, 2022-09-05:
New Alder-Lake based (12th Gen Intel) are starting to appear. Not sure what would be required to support them on full BIOS, but I guess is not a minor work. @MrChromebox  Is that on your plans?

Thanks !!!

MrChromebox, 2022-09-06:
no plans to support them at this time

maxiauer, 2022-09-22:
Just to add further context before closing this, in case other people are interested (and might be able to help). Is it because you do not have the device for testing? If somebody would be willing to test whatever alpha/beta you create (at their own risk) would this be doable? Or is about Tianocore (or some other 3rd party code that you need) not supporting these CPUs yet?

MattDeVillier-AMD, 2022-09-22:
no need to close, I'll eventually add it at some point, just need time + dev hardware. I can certainly package Tianocore up as an AtlFw update and if it works great, but would prefer to be able to test/debug myself before releasing

MrChromebox, 2022-09-22:
no need to close, I'll eventually add it at some point, just need time + dev hardware. I can certainly package Tianocore up as an AtlFw update and if it works great, but would prefer to be able to test/debug myself before releasing

mradenovic, 2022-12-22:
I have a Framework  Laptop Chromebook Edition (banshee). I tried to [build](https://github.com/tianocore/edk2/blob/master/UefiPayloadPkg/BuildAndIntegrationInstructions.txt) TianoCore and  [pack](https://chromium.googlesource.com/chromiumos/docs/+/HEAD/developer_mode.md#alt-firmware) as altfw. 

It doesn't work but I'm not confident I did everything right. If I understood well, UEFI Payload should not be device dependent. Some guidance on how to properly do it would be appreciated, packaged binary would work too.

MrChromebox, 2022-12-22:
> It doesn't work

depthcharge is broken for loading Altfw on BANSHEE

> If I understood well, UEFI Payload should not be device dependent

it's mostly device agnostic, but it depends on the underlying coreboot firmware being sane

>  Some guidance on how to properly do it would be appreciated, packaged binary would work too.

Google needs to fix depthcharge before any payload will boot on BANSHEE (and presumably, all other ADL boards too)

dragon788, 2023-02-20:
@MrChromebox has Google put any effort into fixing this or do we need to file some CrBugs to vote on?

I found it really interesting that in the Alternative Bootloader menu it shows U-Boot as an option, which of course fails, but I'm guessing that is the one that would need replaced or supplemented with a Tianocore based AltFW?

MrChromebox, 2023-02-20:
the problem is that *any* AltFw payload fails to boot. There's a depthcharge or libpayload bug that mishandles the payload and thinks it's an unknown type. I don't know if it's fixed in ToT or in the brya firmwre branches

happybrahma, 2023-05-09:
I have an alder lake OSIRIS board chromebook. Acer 516 GE chromebook.

I'd be happy to test out an alt firmware binary if anybody has built one.. I'm sure I can figure out how to flash it. I'd even be willing to build it myself with some noob-friendly instructions. I have a 20 year development background, but I have never really messed around much with this platform, so I've got a little learning curve to get through.  Looks like there may be a few different things to build and string together.

happybrahma, 2023-05-09:
Probably worth mentioning that I'm just trying to boot grub/ubuntu LTS 22.04.2

MrChromebox, 2023-05-09:
RW_LEGACY firmware for ADL has been added

closed 2023-05-09 by MrChromebox

happybrahma, 2023-05-10:
Successful installed the firmware (edk2), booted USB for ubuntu live install media, installed, and am successfully booting grub and ubuntu 22.4.2 on internal drive partition p7.

Sound does not work on ubuntu, but this probably a ubuntu problem.  I had similar sound problems with tiger lake (Acer Chromebook cb515-1wt)

MrChromebox, 2023-05-10:
even if audio was a firmware issue, there's nothing I can do about it via RW_LEGACY firmware - it's just a bootloader. You're still running Google's broken firmware.

happybrahma, 2023-05-10:
Got it.  Thank you for the support!

maxiauer, 2023-05-11:
> RW_LEGACY firmware for ADL has been added

Thanks a lot MrChromebox !!!

Quick question: any plan for FULL ROM (UEFI Firmware) support?

MrChromebox, 2023-05-11:
> Quick question: any plan for FULL ROM (UEFI Firmware) support?

preliminary support for some boards will be included in the upcoming 4.20 release