MrChromebox/firmware #383: Boot Menu/Settings appearing in Windows Recovery Settings as UiApp

PhantomTimed, 2022-09-10:
When going into Windows Recovery Settings and choosing Use a device, the BIOS appears there as UiApp

MrChromebox, 2022-09-10:
what exactly is the problem here?

PhantomTimed, 2022-09-10:
you go into windows recovery (the place where the troubleshooting options are) and then press use a device the boot menu/settings is there and appears as UiApp

MrChromebox, 2022-09-10:
and what exactly is the problem?

PhantomTimed, 2022-09-10:
nothing, its just weird theres a second option for getting into the uefi firmware settings (going to advanced options shows an option to go to boot menu/settings too, just more obvious on what it does)

MrChromebox, 2022-09-11:
I don't use Windows, have no idea what is or isn't in the recovery options menu, so you'll forgive me for trying to understand what the problem is. I'm certain this is either a Windows bug or something with upstream edk2, so don't plan to spend any time on what's simply a cosmetic issue

closed 2023-01-13 by MrChromebox