MrChromebox/firmware #385: Can't enroll after installing Cloudready on HP Chromebook 14 G4

gabbott507, 2022-09-13:
I have successfully loaded an updated OS on my old EOL chromebook but cannot enroll it to a Google Workspace domain. When I try to do so I get "Enrollment Error, Oops! The system failed to establish the device installation-time attributes lock."

Any help would be appreciated, i have about 20 more i'd like to get working.

MrChromebox, 2022-09-13:
using what firmware? UEFI Full ROM or RW_LEGACY?

gabbott507, 2022-09-13:
> using what firmware? UEFI Full ROM or RW_LEGACY?

UEFI Full ROM

MrChromebox, 2022-09-14:
no idea offhand. Have you cleared/reset the TPM as per my FAQ?

gabbott507, 2022-09-15:
Yes I tried that. Though I'm not sure this chromebook has a tpm chip

gabbott507, 2022-09-27:
It does have a TPM. I tried resetting it again. I can reset it just fine and when I boot into Flex, and go to enrollment, it detects the TPM and tells me to reboot so it can take ownership of it. I do so and when I try to enroll, i still get the error message.

MrChromebox, 2022-09-28:
sorry, I'm not sure what else to suggest. The TPM implementation in Tianocore/edk2 is incomplete currently.

closed 2023-05-17 by MrChromebox