MrChromebox/firmware #386: Tiano Bootloader inaccessible on Lenovo Chromebook

Gitspud2019, 2022-09-17:
Hi Team, today could be the day you save a life!  Not really, but I am 'dying' to solve this issue.  I have a Lenovo Chromebook, a new model, 2022, and have followed many online how-to's for booting from USB... HOWEVER.... when the boot options come up, my 2 button apparently doesn't work and I can't select the Tiano option.  Follow-up question, can I boot Tails uing Tiano?

Gitspud2019, 2022-09-17:
Also, sorry if I'm posting in the wrong place within the big bad internet.  I am a software implementation specialist and system administrator; I just suck at forums because I'm old :)

MrChromebox, 2023-01-13:
closing due to lack of response

closed 2023-01-13 by MrChromebox

Gitspud2019, 2023-01-14:
Sorry for lack of response I actually did get this working by following
your instructions more closely and not rushing.