MrChromebox/firmware #387: How to recover? Asus Chromebox i7 panther with SeaBIOS

wicadmin, 2022-09-19:
Hello,
I have a  Asus Chromebox i7 panther and no way to get to a shell. I tried to revert to chromeos but once that recovery completed due to SeaBIOS being there (what I suspect), it boots to a black screen.

I tried putting other OSes on USB but the keyboard would work to select anything to install them. Same with the Esc for SeaBIOS, it boots straight to HD or USB.

Is there some method to get a shell to start over installing the firmware properly?

MrChromebox, 2022-09-19:
CTRL+D before SeaBIOS loads to boot ChromeOS. Stock SeaBIOS is broken AF and won't recognize any USB (keyboard) at all. Then flash UEFI firmware using my script

PS - you didn't need to restore ChromeOS, that created unnecessary complication.

wicadmin, 2022-09-19:
You know, I could have swore I did that a hundred times yesterday. Anyway, thank you. I've now loaded into ChormeOS. So I don't screw this up any further, what would be my next step? Trying to get LibreElec installed.

wicadmin, 2022-09-19:
Well, keyboard still not working and I'm at a Google Enterprise Enrollment screen asking for an email to be entered.

wicadmin, 2022-09-19:
Plugged in another keyboard, that now works. Still cannot get to a shell.

wicadmin, 2022-09-19:
ok, finally got to a localhost login with Ctrl-Alt-F1

MrChromebox, 2022-09-20:
so all good now? just flash the UEFI firmware, then reboot and install LibreELEC from USB/SD media (preferably created by their install tool)

closed 2023-01-13 by MrChromebox