MrChromebox/firmware #391: Latitude 7410 Chromebook // RW_LEGACY Firmware / UEFI Firmware (Full ROM)

Qwasu1987, 2022-09-26:
Hi all,

Is there any reasonable way of installing WINDOWS OS or any other OS than CHROME on **Latitude 7410 Chromebook**
I have run through crush command provided
 1 for RW_LEGACY Firmware  
 2 for UEFI Firmware (Full ROM)
 3 & 4  I could not access.
 
Is this cause because of Write Protect is Enabled ?
How I can turn off WP as I have run it in ADMIN mode and the bit in the below link
https://mrchromebox.tech/#devices 
Seems a little complicated to me.
Is there any easier way of doing it ? Is someone in position to guide me a little through the process ?

Thank you for your help in this matter :)

MrChromebox, 2022-09-26:
I'm not sure how I can make my Supported Devices page more clear - DRALLION does not have either RW_LEGACY nor UEFI Full ROM firmware available currently. It has nothing to do with the firmware write protection.

I can't remember why I don't offer RW_LEGACY - it may simply fail to boot. I don't offer a UEFI Full ROM because there's no easy way for most users to recover from a bad flash - it doesn't support CR50 reprogramming, and the flash chip is not clippable. If you're undaunted by that, I can provide you a custom firmware build to test. It won't run Windows though and USB-C ports won't work under Linux (except for charging -- no data nor alt-mode functions)

MrChromebox, 2022-09-26:
I have a DRALLION here. RW_LEGACY doesn't boot (tried a fresh build today). With UEFI Full ROM firmware, Windows doesn't boot without substantial hackery. USB-C ports don't work. TB doesn't work. No built-in audio/mic/HP jack. I didn't do extensive testing, but it's not really usable. Just stick with ChromeOS IMO

MrChromebox, 2022-11-30:
DRALLION now has both RW_LEGACY and UEFI Full ROM firmware available. Windows is bootable, but the battery ACPI is still a bit buggy so not showing under Windows. If someone loans me a DRALLION I can probably fix it, but it's not a high priority at the moment

closed 2022-11-30 by MrChromebox