MrChromebox/firmware #395: Request: Toggle for auto power-on when AC power ON

carstentrink, 2022-10-01:
Add Toggle in firmware to auto power-on when AC power
Reference:
https://old.reddit.com/r/chrultrabook/comments/xqbp4s

(Thx)

Copyrites, 2023-02-26:
This is a feature that would be useful to me too.  My system is an Asus CN60 M004U.  Thank you

junnikokuki, 2023-05-24:
I would like not to power on when lid open on lillipup.

MrChromebox, 2023-05-24:
@junnikokuki your request is completely different than this issue, search the issues list as your request has already been made

carstentrink, 2023-08-10:
Could you please give some test image for this please?

- Electricity prices in EU are getting very high
- At the moment, I am having only SNAPPY and DROID

Thank you!

MrChromebox, 2023-08-10:
@carstentrink for what exactly? No idea what high electric costs have to do with powering on when you open the lid

carstentrink, 2023-08-13:
@MrChromebox  I meant. Auto powerON when AC (i.e) USB-C charger is connected. (see comment in reddit or first comment)

MrChromebox, 2023-08-13:
@carstentrink it should without a battery connected, but if not then you're looking at custom EC and system firmware

carstentrink, 2023-08-13:
Oh.. I never thought of trying that way. At the moment the battery serves as UPS for running a` shutdown -h now `gracefully after some hours.

closed 2023-08-13 by carstentrink