MrChromebox/firmware #396: Anybody flashed full firmware to Acer Chromebook 317? Don't want to brick this.

wesmorrison, 2022-10-04:
So far, I've been using updated RW_LEGACY so I can dual boot ChromeOS as a backup, but I really have no need for ChromeOS and would rather use the full drive for Linux.

But once I do, there's no going back to the original firmware, right? The only reason I'd want to is if the full firmware didn't work, but then, if I brick it...

Should I hold off flashing full firmware to my Acer Chromebook 317? (As I understand, to disable WP, I poweroff, disconnect power supply, open case, disconnect battery, reconnect power supply, boot, flash, shut down, reconnect battery and close case.)

MrChromebox, 2022-10-04:
there's no UEFI Full ROM firmware for your device as per https://mrchromebox.tech/#devices

wesmorrison, 2022-10-04:
Thanks. UEFI Full ROM firmware for my device doesn't exist. Does that mean:

- It's not possible
- You need the device to even look into it
- Too busy, not interested. :)

MrChromebox, 2022-10-05:
mix of #2 and #3. Also, Jasperlake Chromebooks don't disable WP with the battery disconnected, so need to figure out the correct pads to jumper to disable it

MrChromebox, 2022-11-30:
UEFI Full ROM support added in 4.18.2 release :)

closed 2022-11-30 by MrChromebox