MrChromebox/firmware #397: UEFI Support for "IdeaPad Flex 5i Chromebook" (LILLIPUP-MQUZ)

Rampaging-Mongoose, 2022-10-04:
Was curious if full UEFI would be possible for this device. I have no qualms with bricking/unbricking the device as long as unbricking methods are outlined.

MrChromebox, 2022-10-04:
possible? sure. unbricking will require a homemade Suzy-Q cable though since NLA for purchase

Rampaging-Mongoose, 2022-10-04:
Guess I should be more direct: what can I do to contribute to actually getting full UEFI on the device?

MattDeVillier-AMD, 2022-10-04:
build a Suzy-Q cable and email me when ready for a test build

Rampaging-Mongoose, 2022-10-13:
> build a Suzy-Q cable and email me when ready for a test build

I ordered this (https://www.digikey.com/en/products/detail/saiko-systems-ltd/brk-usb-cpv3-0/15283090) which should be arriving today and I think I have the necessary resistors lying around. With any luck I should be able to build and test the cable tomorrow.

MrChromebox, 2022-10-13:
once you have it built, as long as you can access the ttyUSB0/1/2 consoles with it, then should be good to go

Rampaging-Mongoose, 2022-10-15:
Alright, got ttyUSB0/1/2.

MrChromebox, 2022-11-30:
added in 4.18.2 release :)

closed 2022-11-30 by MrChromebox

marcosilva85, 2022-12-01:
> 
Great! I wonder how stable and safe it is to use it?
Is there anything that can still improve?

@Rampaging-Mongoose if you have already tried it, how is your experience?

MrChromebox, 2022-12-01:
only firmware-related issue with TGL right now is that the TB/USB-C ports don't work for booting, but that's being worked on

marcosilva85, 2022-12-01:
That seems that is at least better since the RW_Legacy the Usb-C port was not responding (fedora Linux).

I tried the Fw utility script on also a TGL Lillipup - option 2 shows greyed out:
[image]

Rampaging-Mongoose, 2022-12-01:
> > 
> 
> Great! I wonder how stable and safe it is to flash and use the full-ROM (I saw it was closed-beta before, so I wondered why)? Is there anything still to improve?
> 
> @Rampaging-Mongoose if you have already tried it, how is your experience? (Touchscreen,Usb-C, battery usage,audio?)

Touch still doesn't work.

So far I have not successfully installed any linux distros. There may be an issue with the eMMC? I'll try to install ubuntu and see what happens.

Rampaging-Mongoose, 2022-12-01:
`Booting from "ubuntu" failed: verify it contains a 64-bit UEFI OS.`

A similar problem happens with all linux distros I try to install. They either throw an error message like that or fail to even attempt booting.

MrChromebox, 2022-12-01:
@marcosilva85 just pushed a fix, these hypenated HWIDs are causing some issues

@Rampaging-Mongoose touchscreens may need power sequencing implemented since they rely on a kernel detection mechanism not implemented in upstream Linux. Audio requires kernel/driver support and SOF blobs/firmware.

marcosilva85, 2022-12-02:
ok, got it to flash now and now into the full-ROM I booted fedora linux (which I used before with RW_LEGACY)
Here is an status report:
**usb-c still not working**
- Its detection seems to actually get worse as now in "dmesg" I don't even see the errors about it anymore: cros-ec-typec.

**audio not detected or working anymore.  (regression)**:
-  I had working audio before, by installing alsa-sof-firmware, but now no reference on dmesg either about it. already tried to manually load the module sof-audio-pci-intel-tgl and also to restart it, with no effect.

marcosilva85, 2022-12-02:
> a) sounds like the cros-ec driver isn't loading, probably due to idiotic DMI string matching b) possibly same as above c) power sequencing in coreboot may fix this

is there any fix or workaround, or something in Linux I could do try and get sound&usb working?

marcosilva85, 2022-12-16:
@MrChromebox,  from your side is there anything that could help us getting sound&usb working?

MrChromebox, 2022-12-16:
@marcosilva85 audio isn't a firmware issue. USB-C not working isn't a firmware issue outside of the UEFI bootloader itself. If crosec-typec driver isn't working, then that's why your USB-C/TB ports aren't

marcosilva85, 2022-12-19:
> @marcosilva85 audio isn't a firmware issue. USB-C not working isn't a firmware issue outside of the UEFI bootloader itself. If crosec-typec driver isn't working, then that's why your USB-C/TB ports aren't

@MrChromebox 
I'm not sure if you got my question. On point b) I said basically audio was working well in Linux before. Just after I've migrated to Full-ROM  **there is some regression, as it is not being detected now**

That is why I'm asking if you have any idea or tips on how to diagnose or fix this.
Could you perhaps explain what your previous comment means on the "DMI string matching" and what could a user like me do?
Or is it something that can only be fixed now on the ROM side?
Thanks

MrChromebox, 2022-12-19:
to try and diagnose, would need the dmesg log from booting Linux with both the UEFI Full ROM and RW_LEGACY firmware, possibly with SoF debug output on. Also cbmem log from both firmwares

marcosilva85, 2023-02-11:
invalid closing reason - issue still active. (Issue could be reopened) or just follow-up in new issue: #430

Rampaging-Mongoose, 2023-02-28:
@MrChromebox Okay finally zeroed in on an issue I've had since I first tried your BIOS. It appears that SD cards are not being recognized by the BIOS (I tried to use one as a boot device) and whenever I try to boot from mmc after installing an OS (I've tried debian-testing, KDE-Neon, and many others with newer kernels) I get the following error:

```
Booting from 'neon' failed; verify it contains a 64-bit UEFI OS.

Press any key to continue booting...
```

Booting from USB is okay but nothing else will boot.

MrChromebox, 2023-02-28:
what's the 'neon' bootorder entry pointing to? does that path/file exist?