MrChromebox/firmware #399: Morphius RW_LEGACY Firmware cannot boot from eMMC or MicroSD

jacob019, 2022-10-05:
I have the Lenovo Chromebook C13.  The RW_LEGACY firmware cannot boot from eMMC or MicroSD, only from USB.

```
02:00.0 SD Host controller: Genesys Logic, Inc GL9750 SD Host Controller (rev 01) (prog-if 01)
	Subsystem: Genesys Logic, Inc GL9750 SD Host Controller
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 31
	IOMMU group: 0
	Region 0: Memory at e0700000 (32-bit, non-prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel driver in use: sdhci-pci
	Kernel modules: sdhci_pci
```

MrChromebox, 2022-10-05:
known issue, the eMMC and SD card are memory mapped devices, Tianocore has no driver for handle that -- only PCI devices.

jacob019, 2022-10-05:
Thanks for the quick response friend.  I guess I'll use Eupnea to boot Linux on this device.  I appreciate you.

MrChromebox, 2024-04-28:
this has been fixed

closed 2024-04-28 by MrChromebox

v-dejan, 2024-05-23:
I am having this same issue on same model of laptop. I don't get it how is it fixed, maybe I am missing something. I installed your script a week ago, booting from the USB stick works perfectly. Because I have small hard drive I decided to install ubuntu on SD card. But there is no SD card in the boot menu. I tried inserting SD card after getting to the boot menu, it doesn't work. It works for the USB. I also tried enabling some flags that allow Chrome OS to be booted from external devices. But, that didn't help either. If you have step by step solution to this issue or if I am missing something I would really appreciate any help. Thanks.

MrChromebox, 2024-05-23:
>  I tried inserting SD card after getting to the boot menu, it doesn't work.

it's not going to

> It works for the USB.

SD and USB drivers work differently

>  I also tried enabling some flags that allow Chrome OS to be booted from external devices.

you're not booting ChromeOS from an external device

>  I don't get it how is it fixed

I uploaded a new RW_LEGACY file which uses coolstar's eMMC driver for edk2. I don't have a eMMC-enabled device to test, but it's been tested successfully with the UEFI Full ROM so it should work for RWL as well. If it's not, then will have to figure out what's wrong