MrChromebox/firmware #401: Akemi upgrade soldered ram

xinghot2007, 2022-10-27:
I have upgraded my Chromebook (ideapad flex 5 13iml05) soldered 4GB RAM to 8GB RAM.(my device havs any larger RAM version)

I soldered the same original DDR4 modules(SAMSUNG K4A8G165WC BCWE) on the empty solder pad and changed MEM_CH_SEL to dual channel(solder R678 to R682).

chrome://system shows that channel 1 have nothing,the system cannot recognize the new 4GB RAM.So i flashed the UEFI firmware use your Firmware Utility Script.The chromebook cannot turn on.When i recover the MEM_CH_SEL to singer channel,the chromebook can boot.

My MEMORY CONFIG:
MEM_STRAP_0 = 0
MEM_STRAP_1 = 1
MEM_STRAP_2 = 1
MEM_STRAP_3 = 0
[https://github.com/MrChromebox/coreboot/blob/2022.07.08/src/mainboard/google/hatch/variants/akemi/Makefile.inc](url)
it's possible that firmware select the right SPD.

I doubt the DDR4 modules which i soldered were damaged, so i buy some new modules and soldered again.The problem is still.

I find this file:
[https://github.com/coreboot/coreboot/blob/master/src/mainboard/google/hatch/variants/baseboard/memory.c](url)
Rcomp resistors(R2 R3 R4) on my motherboard is 121 81 100,but i don`t understand the rest of code:
        /* Baseboard Rcomp target values */
	.rcomp_targets = {100, 40, 20, 20, 26},

	/* Set CaVref config to 2 */ 
	.vref_ca_config = 2,

	/* Enable Early Command Training */
	.ect = 1,
How to measure the rcomp target values?
How can the motherboard recognize my memory?

Thank you！

MrChromebox, 2022-10-27:
the rcomp values are based on trace lengths on the board, they do no change based on memory type/amount installed.

to troubleshoot this, I would recommend a debug build of coreboot and a Suzy-Q cable so you can see why ram init is failing / see how the firmware is recognizing the installed memory and strap config

xinghot2007, 2022-10-29:
Thank you for your advice!

I made a Suzy-Q cable and ccd CMD "version" show:

> version 
Chip:    g cr50 B2-C
Board:   0
RO_A:    0.0.10/29d77172
RO_B:  * 0.0.11/4d655eab
RW_A:  * 0.6.120/cr50_v3.94_pp.43-adfaeb3b2e
RW_B:    Error
BID A:   46464646:00000000:00000010 Yes
BID B:   46464646:00000000:00000010 Yes
Build:   0.6.120/cr50_v3.94_pp.43-adfaeb3b2e
         tpm2:v1.9308_26_0.71-8b56d7c
         2022-06-06 21:06:11 @chromeos-ci-firmware-us-east1-d-x32-0-n9s3

Is means the RW_B have some ERRORS?

When use make crossgcc,which suffix to use? the -i386 or the -x64.

when use make menuconfig,I can`t decision which options to enable.Especially the EC,my board is NPCX796.

when use make,the console show:

- Done -
Build end time: 01:16:07, Oct.29 2022
Build total time: 00:03:02

Success!
    CC         romstage/mainboard/google/hatch/static.o
    CC         romstage/acpi/acpi_pm.o
    CC         romstage/arch/x86/assembly_entry.o
    CC         romstage/arch/x86/boot.o
    CC         romstage/arch/x86/breakpoint.o
    CP         romstage/arch/x86/car.ld
    CC         romstage/arch/x86/cf9_reset.o
    CC         romstage/arch/x86/cpu_common.o
    CC         romstage/arch/x86/exception.o
    CC         romstage/arch/x86/gdt_init.o
    CC         romstage/arch/x86/idt.o
    CC         romstage/arch/x86/memcpy.o
    CP         romstage/arch/x86/memlayout.ld
    CC         romstage/arch/x86/memmove.o
    CC         romstage/arch/x86/memset.o
    CC         romstage/arch/x86/mmap_boot.o
    CC         romstage/arch/x86/null_breakpoint.o
    CC         romstage/arch/x86/post.o
    CC         romstage/arch/x86/postcar_loader.o
    CC         romstage/arch/x86/romstage.o
    CC         romstage/arch/x86/timestamp.o
    CC         romstage/commonlib/bsd/cbfs_mcache.o
    CC         romstage/commonlib/bsd/cbfs_private.o
    CC         romstage/commonlib/bsd/elog.o
    CC         romstage/commonlib/bsd/lz4_wrapper.o
    CC         romstage/commonlib/iobuf.o
    CC         romstage/commonlib/mem_pool.o
    CC         romstage/commonlib/rational.o
    CC         romstage/commonlib/region.o
    CC         romstage/console/console.o
    CC         romstage/console/die.o
    CC         romstage/console/init.o
    CC         romstage/console/post.o
    CC         romstage/console/printk.o
    CC         romstage/console/vsprintf.o
    CC         romstage/console/vtxprintf.o
    CC         romstage/cpu/intel/car/romstage.o
    CC         romstage/cpu/intel/microcode/microcode.o
    CC         romstage/cpu/x86/lapic/boot_cpu.o
    CC         romstage/cpu/x86/lapic/lapic.o
[... 149 lines cut ...]
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: build/bootblock/arch/x86/exception.o: in function `stub_putc':
/home/xinghot2007/coreboot/src/arch/x86/exception.c:227: undefined reference to `gdb_tx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: /home/xinghot2007/coreboot/src/arch/x86/exception.c:227: 
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: /home/xinghot2007/coreboot/src/arch/x86/exception.c:227: undefined reference to `gdb_tx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: /home/xinghot2007/coreboot/src/arch/x86/exception.c:227: undefined reference to `gdb_tx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: /home/xinghot2007/coreboot/src/arch/x86/exception.c:227: undefined reference to `gdb_tx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: build/bootblock/arch/x86/exception.o: in function `stub_flush':
/home/xinghot2007/coreboot/src/arch/x86/exception.c:232: undefined reference to `gdb_tx_flush'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: build/bootblock/arch/x86/exception.o: in function `stub_getc':
/home/xinghot2007/coreboot/src/arch/x86/exception.c:237: undefined reference to `gdb_rx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: /home/xinghot2007/coreboot/src/arch/x86/exception.c:237: undefined reference to `gdb_rx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: /home/xinghot2007/coreboot/src/arch/x86/exception.c:237: undefined reference to `gdb_rx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: /home/xinghot2007/coreboot/src/arch/x86/exception.c:237: undefined reference to `gdb_rx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: /home/xinghot2007/coreboot/src/arch/x86/exception.c:237: undefined reference to `gdb_rx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: build/bootblock/arch/x86/exception.o: in function `stub_putc':
/home/xinghot2007/coreboot/src/arch/x86/exception.c:227: undefined reference to `gdb_tx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: build/bootblock/arch/x86/exception.o: in function `stub_flush':
/home/xinghot2007/coreboot/src/arch/x86/exception.c:232: undefined reference to `gdb_tx_flush'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: build/bootblock/arch/x86/exception.o: in function `stub_putc':
/home/xinghot2007/coreboot/src/arch/x86/exception.c:227: undefined reference to `gdb_tx_byte'
/home/xinghot2007/coreboot/util/crossgcc/xgcc/bin/i386-elf-ld.bfd: build/bootblock/arch/x86/exception.o: in function `stub_flush':
/home/xinghot2007/coreboot/src/arch/x86/exception.c:232: undefined reference to `gdb_tx_flush'
make: *** [src/arch/x86/Makefile.inc:103: build/cbfs/fallback/bootblock.debug] Error 1

I dont know whether it succeed or fail?

MrChromebox, 2022-10-29:
I'd recommend cloning my repo, ensuring you have all required packages to build installed, then running `make crossgcc-i386` then edit `configs/cml/config.akemi.uefi` to remove the line disabling serial output. save, then build by running `./build-ueif.sh akemi`

xinghot2007, 2022-10-30:
Under your advice,I cloned your repo.but I cannot find the line "**disabling serial output**" in _configs/cml/config.akemi.uefi_.
So I add two lines to _configs/cml/config.akemi.uefi_ like below:
CONFIG_CONSOLE_SERIAL=y
CONFIG_DEFAULT_CONSOLE_LOGLEVEL=8
I don`t hnow if that`s right.
The debug msg:
[NOTE ]  coreboot-4.14-7726-g8a76e170c6-dirty-MrChromebox-4.18.1 Thu Oct 27 13:24:50 UTC .
[DEBUG]  pm1_sts: 0900 pm1_en: 0000 pm1_cnt: 00001c00
[DEBUG]  gpe0_sts[0]: 00000000 gpe0_en[0]: 00000000
[DEBUG]  gpe0_sts[1]: 00200000 gpe0_en[1]: 00000000
[DEBUG]  gpe0_sts[2]: 00000000 gpe0_en[2]: 00000000
[DEBUG]  gpe0_sts[3]: 00000000 gpe0_en[3]: 00000000
[DEBUG]  TCO_STS:   0000 0000
[DEBUG]  GEN_PMCON: a0054000 00000204
[DEBUG]  GBLRST_CAUSE: 00000000 00000000
[DEBUG]  HPR_CAUSE0: 00000000
[DEBUG]  prev_sleep_state 5
[DEBUG]  FMAP: area COREBOOT found @ c04200 (4177408 bytes)
[INFO ]  CBFS: Found 'fspm.bin' @0x8bdc0 size 0x8e000 in mcache @0xfef21688
[DEBUG]  FMAP: area RW_MRC_CACHE found @ bb0000 (65536 bytes)
[NOTE ]  MRC: no data in 'RW_MRC_CACHE'
[DEBUG]  SPD INDEX = 6
[INFO ]  CBFS: Found 'spd.bin' @0x60180 size 0xe00 in mcache @0xfef215c8
[INFO ]  SPD: module type is DDR4
[INFO ]  SPD: banks 8, ranks 1, rows 16, columns 10, density 8192 Mb
[INFO ]  SPD: device width 16 bits, bus width 64 bits
[INFO ]  SPD: module size is 4096 MB (per channel)
[INFO ]  memory slot: 0 configuration done.
[INFO ]  memory slot: 2 configuration done.
[EMERG]  FspMemoryInit returned with error 0x80000007!

MrChromebox, 2022-10-30:
> Under your advice,I cloned your repo.but I cannot find the line "disabling serial output" in configs/cml/config.akemi.uefi.

simply remove the line:
> # CONFIG_CONSOLE_SERIAL is not set

> [DEBUG] SPD INDEX = 6
that's `SPD_SOURCES += 8G_3200		# 0b110`

this doesn't seem correct. That selection is for the module, not the total capacity. Selecting index 6 is correct for 4GB of DDR4-3200 RAM (I know this bc my 4GB AKEMI has index 6). What was the original SPD index/strapping of your board?

xinghot2007, 2022-11-04:
My original SPD index is 6(use single channel).It is means bit not byte.

According to the schematic diagram,the motherboard DQMAP is below:

CHIP_NO       DQSL_T                   DQSL_C                       DQSU_T                DQSU_C
1.                    DDR_A_DQS1         DDR_A_DQS#1           DDR_A_DQS2         DDR_A_DQS#2
2.                    DDR_A_DQS3         DDR_A_DQS#3           DDR_A_DQS0         DDR_A_DQS#0
3.                    DDR_A_DQS6         DDR_A_DQS#6           DDR_A_DQS4         DDR_A_DQS#4
4.                    DDR_A_DQS5         DDR_A_DQS#5           DDR_A_DQS7         DDR_A_DQS#7
5.                    DDR_B_DQS0         DDR_B_DQS#0           DDR_B_DQS1         DDR_B_DQS#1
6.                    DDR_B_DQS4         DDR_B_DQS#4           DDR_B_DQS2         DDR_B_DQS#2
7.                    DDR_B_DQS5         DDR_B_DQS#5           DDR_B_DQS3         DDR_B_DQS#3
8.                    DDR_B_DQS6         DDR_B_DQS#6           DDR_B_DQS7         DDR_B_DQS#7

[https://github.com/MrChromebox/coreboot/blob/2022.10.24/src/soc/intel/cannonlake/include/soc/cnl_memcfg_init.h](url)
This cnl_mb_cfg struct shows dqs_map is only used by LPDDR.

Whether or not to define the dqs_map in this file:
[https://github.com/MrChromebox/coreboot/blob/2022.10.24/src/mainboard/google/hatch/variants/baseboard/memory.c](url)

[NOTE ] MRC: no data in 'RW_MRC_CACHE'
Whether or not it means the temp memory not init in FSP?

Thank you and best wishes!

xinghot2007, 2022-11-04:
CHIP_NO     DQSL_T      DQSL_C               DQSU_T                DQSU_C

1.       DDR0_DQSP1      DDR0_DQSN1      DDR0_DQSP2      DDR0_DQSN2
2.       DDR0_DQSP3      DDR0_DQSN3      DDR0_DQSP0      DDR0_DQSN0
3.       DDR0_DQSP6      DDR0_DQSN6      DDR0_DQSP4      DDR0_DQSN4
4.       DDR0_DQSP5      DDR0_DQSN5      DDR0_DQSP7      DDR0_DQSN7
5.       DDR1_DQSP0      DDR1_DQSN0      DDR1_DQSP1      DDR1_DQSN1
6.       DDR1_DQSP4      DDR1_DQSN4      DDR1_DQSP2      DDR1_DQSN2
7.       DDR1_DQSP5      DDR1_DQSN5      DDR1_DQSP3      DDR1_DQSN3
8.       DDR1_DQSP6      DDR1_DQSN6      DDR1_DQSP7      DDR1_DQSN7

MrChromebox, 2022-11-04:
> My original SPD index is 6(use single channel).It is means bit not byte.

I understand how the GPIO straps work.

> [NOTE ] MRC: no data in 'RW_MRC_CACHE'
> Whether or not it means the temp memory not init in FSP?

this means that RAM training was not successfully completed / that there is no cached RAM training data in the RW_MRC_CACHE region of the flash chip. This is expected after flashing the firmware until you successfully boot.

what is GPIO GPP_F2 (MEM_CH_SEL) set to? 0 (dual channel) or 1 (single channel)?

I'm fairly certain this isn't a DQ map issue.

xinghot2007, 2022-11-05:
The GPIO GPP_F2 has connected to the ground with a 100k ohm resistor.(solder R679 to R682)
I have already booked some micron’s ram chips for replace,because I found the 8GB version motherboard use these.I hope it works.
Thank you and best wishes!

MrChromebox, 2022-11-05:
was that something you added or already there? If the latter, could be why you weren't able to add more

closed 2026-11-04 by MrChromebox