MrChromebox/firmware #402: Nocturne 1027 UEFI full ROM not accepting keyboard input from the slate keyboard, and not recognizing usb keyboards.

yw662, 2022-10-31:
( It is technically a brick for me now since I have currently no way to connect CCD, sad story. )
If I close/open the lid, the slate will shutdown/boot automatically, but I can do nothing at the UEFI firmware interface since it is not accepting keyboard input.
It can charge, but it refuse to recognize my USB keyboard. And it refuse to automatically boot from a USB disk, I have no idea whether the disk itself is recognized.
If I can get CCD work I would hopefully be able to flash via CCD, but is it a firmware issue or it can just be about nvram ?

yw662, 2022-10-31:
Or maybe I can just ship it to you. Did you just say you don't currently have a nocturne in hand ?

olm3ca, 2022-11-26:
@MrChromebox is this an ongoing issue? I have a NOCTURNE and want to use UEFI but not if it's going to cause this same problem. Thanks

MrChromebox, 2022-11-26:
it's a problem with the current 4.18.1 firmware. Got a Google keyboard and can reproduce. it's being worked on

MrChromebox, 2022-11-30:
fixed in 4.18.2 release :)

closed 2022-11-30 by MrChromebox

yw662, 2022-11-30:
So, technically, how should I flash the new version for my nocturne ?

MrChromebox, 2022-11-30:
* boot Linux, run script
* boot ChromeOS, run script
* flash directly using Suzy-Q cable