MrChromebox/firmware #405: ASUS C214 CHROMEBOOK - Issue booting into USB

georgekrds, 2022-11-03:
**Hello I can't boot into usb (it says: error: file `/syslinux/vmlinuz.A' not found). When the laptop boots, I press _ctrl+L_ and it boots me to the boot manager that I downloaded for mr'chromebox.**

### **Harware Info:**
_Official Website:_ **https://www.asus.com/laptops/for-home/chromebook/asus-chromebook-flip-c214/techspec/**
_Processor:_ Intel® Celeron® N4120 Processor 1.1 GHz (4M Cache, up to 2.6 GHz, 4 cores)
_Graphics:_ Intel® UHD Graphics 600
_Memory:_ 8G LPDDR4 on board
_Storage:_ 64G eMMC
_I/O Ports:_ 1x USB 3.2 Gen 1 Type-A | 2x USB 3.2 Gen 1 Type-C support display / power delivery | 1x 3.5mm Combo Audio Jack | Micro SD card reader

### **System Info:**
_Google Chrome:_ 107.0.5304.92 (Official Build) (64-bit)
_Revision:_ ad1ccc3c518603d258a534f3c03680d4726dc7eb-refs/branch-heads/5304@{#1101}
_Platform:_ 15117.86.0 (Official Build) stable-channel octopus
_Firmware Version:_ Google_Ampton.11297.314.0
_Customization ID:_ ampton
_ARC:_ 9196637 
_SDK:_ Version: 30

### **Photos**
[image]
[image]
[image]
[image]

MrChromebox, 2022-11-03:
there is no problem here, it is trying to boot into ChromeOS because you have not selected to boot from USB.

Press ESC on the boot splash when it says to. Go to Boot Menu. Select your USB and press enter.

closed 2022-11-03 by MrChromebox

georgekrds, 2022-11-03:
I already tried and it gave me: error: disk 'hd0,3' not found

georgekrds, 2022-11-03:
Nevermind I just edited the boot manager. If you would like you can add a text that says something like press esc for boot manager. Thank you very much for your help and your project. Im was trying to to this for quite some time but my device wasn't supported. Thank you for your work it made me very happy! 😁

MrChromebox, 2022-11-04:
It literally says "Press ESC for boot options/settings" on the screenshot you posted above :)