MrChromebox/firmware #406: Toshiba Chromebook 2 (2014) CB30/CB35 (SWANKY)

ka9314, 2022-11-16:
Hello All,

I has some issue after updated the coreboot(from 4.16 to 4.18.1),

1. My chromebook cannot reboot, it will freeze in the rabbit logo.
2. My usb keyboard is working in the UEFI shell and grub menu, but the usb keyboard and mouse will freeze ~1mins after booting into linux. (The build-in keyboard and touchpad is ok)

Also, please let me know, is there a way to downgrade back to 4.16.
Thanks.

Chromebook: Toshiba Chromebook 2 (2014) CB30/CB35 (SWANKY)
OS: archlinux

MrChromebox, 2022-11-16:
not seeing this issues on SWANKY here, running Pop_OS! 22.04

ka9314, 2022-11-17:
Thank you for the reply,

I tried to reinstall the coreboot or change the linux kernel. but it didn't help.
I found some error messages in the dmesg.

> [   48.735912] usb 1-4: device descriptor read/64, error -110
> [   64.309161] usb 1-4: device descriptor read/64, error -110
> [   64.416067] usb usb1-port4: attempt power cycle
> [   64.819123] usb 1-4: new full-speed USB device number 6 using xhci_hcd
> [   69.945826] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
> [   70.152678] usb 1-4: Device not responding to setup address.
> [   70.359135] usb 1-4: device not accepting address 6, error -71
> [   70.482455] usb 1-4: new full-speed USB device number 7 using xhci_hcd
> [   75.492537] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
> [   75.699477] usb 1-4: Device not responding to setup address.
> [   75.905759] usb 1-4: device not accepting address 7, error -71
> [   75.906042] usb usb1-port4: unable to enumerate USB device

I believe something happened in the usb controller. But I have no idea now.

[dmesg_20220908.txt](https://github.com/MrChromebox/firmware/files/10027481/dmesg_20220908.txt)
[dmesg_20221111.txt](https://github.com/MrChromebox/firmware/files/10027491/dmesg_20221111.txt)
[lspci.txt](https://github.com/MrChromebox/firmware/files/10027520/lspci.txt)
[lsusb.txt](https://github.com/MrChromebox/firmware/files/10027521/lsusb.txt)

MrChromebox, 2022-11-17:
shoot me an email (mrchromebox at gmail) and I'll have you roll back to the previous release (4.17.2) and see if that fixes it

ka9314, 2022-11-17:
Thanks for your help,

After rolling back the coreboot, the issue is fixed.

For your information, 
lsusb shows that the usb 1-4 is the Bluetooth wireless interface.
and my wireless card shall be the original one.

[dmesg.txt](https://github.com/MrChromebox/firmware/files/10034613/dmesg.txt)
[lspci.txt](https://github.com/MrChromebox/firmware/files/10034614/lspci.txt)
[lsusb.txt](https://github.com/MrChromebox/firmware/files/10034615/lsusb.txt)

MrChromebox, 2022-11-30:
please try the 4.18.2 release and let me know if that resolves the issue

ka9314, 2022-12-02:
The new version is working great.

Thanks

MrChromebox, 2022-12-02:
great, thanks for letting me know

closed 2022-12-02 by MrChromebox