MrChromebox/firmware #408: Samsung Galaxy Chromebook 2 360 (BUGZZY)

saurik, 2022-11-26:
Hello! I'd love to see UEFI support for the Samsung Galaxy Chromebook 2 360 (BUGZZY). I'd be happy to provide hardware for this to happen. If it is at all interesting, it claims to have "Fw Ver: Google_Bugzzy.13606.465.0 (06/02/2022)".

MrChromebox, 2022-11-26:
The issue with Jasperlake Chromebooks is the difficulty in disabling firmware WP - need to use a SuzyQ cable or bridge an unmarked jumper on the mainboard. I can add RW_LEGACY/AltFw support for it though

saurik, 2022-11-26:
@MrChromebox FWIW, while I had not realized I would need to do so (as I hadn't known that disconnecting the battery wouldn't work until you mentioned it... I wish I'd realized that before spending forever figuring out how to carefully open the case... lol), I had purchased all of the parts required to build one of those cables this week (because I had expected I'd need them for another device, before I learned I could simply disconnect the battery on that one), and so I went ahead and assembled the cable (which wasn't that difficult, despite my only having Electrical Engineer +0.25). It seems to work and I was successfully able to disable the write protection! I'd thereby be able to: 1) use any resulting final firmware; 2) help test any in-progress firmware; or 3) ship you hardware to work with that is already set to "wp false atboot".

MrChromebox, 2022-11-26:
ok cool, drop me an email (MrChromebox at gmail) next week and I'll whip up a test build for you

MrChromebox, 2022-11-30:
support added in 4.18.2 release :)

closed 2022-11-30 by MrChromebox