MrChromebox/firmware #409: UEFI BIOS not remembering boot

Welarmo, 2022-12-02:
Hi everytime I boot I get the UEFI shell. I have gone ESC on boot changed the boot order to my USB storage as first my internal SSD Linux pop os second then UEFI shell last. F10 saved committed changed rebooted and then brings me back to UEFI shell just looping. I have tried deleting the other noot methods but the just appear back on reboot. I have even tried changing the boot order through my Linux drive in efibootmgr setting the usb 0005 at the start. Success through command line but after reboot back to the UEFI shell. What is going wrong how can I force it to remember my preferred boot order? Thanks

MrChromebox, 2022-12-02:
device? firmware version? tried clearing NVRAM?

walmartshopper, 2022-12-31:
Do you happen to have a model that boots the Tianocore UEFI as a RW_LEGACY payload?  I have a tigerlake and had this issue where I would install grub but the grub entry would be gone after a reboot.  I was able to get it booting by copying the grub efi file to the default fallback path of <EFI Partition>/boot/bootx64.efi.  This is the default path that Tianocore looks for when booting from the internal drive.  So if you set it up that way, you shouldn't need any special boot entries.  By default it should try booting from the internal drive, find bootx64.efi, and boot successfully.  The bootx64.efi file doesn't have to be grub either, any other efi bootloader should work if you copy it to that location.

MrChromebox, 2023-01-13:
closing due to lack of response

closed 2023-01-13 by MrChromebox

nangiloc, 2023-06-16:
This helped me, but the correct location is: EFI_PARTITION/EFI/Boot/bootx64.efi