MrChromebox/firmware #410: Control fan on a Dell Chromebox 3010 (Intel Haswell / TRICKY)

user8446, 2022-12-02:
Hi,

I have a Dell Chromebox 3010 which is an Intel Haswell (TRICKY) Celeron 2955U 2 core 1.4Ghz.

Installed the firmware and loaded Lubuntu. However, when turning on the box the fan immediately turns on and stays at one speed no matter the load.

There was this error on boot: `cros_ec_epcs cros_ec_lpcs.0: EC ID not detected`. After reading [this similar issue](https://github.com/MrChromebox/firmware/issues/200), I added `blacklist cros_ec_lpcs` to `/etc/modprobe.d/blacklist.conf` which did remote the error but no change to the fan.

I installed `lm-sensors` and it can see temps but I don't see fan speed. Does the Chromebox even have fan speed control? If so, does anyone know how to control it?

MrChromebox, 2022-12-03:
as I mentioned in the other issue, pre-2018 model Chromeboxes do not have an embedded controller. The cros-ec driver attempting to load and failing is meaningless (because again, there IS NO EC).

the fan on pre-2018 Chromeboxes is controlled by the ITE IT8772F SuperIO controller chip, and the fan can be interfaced with via the ITE87 driver (which may not be part of the mainline kernel anymore) under Linux, or SpeedFan under Windows.

by default, the fan speed is set based on temperature; you can see the default settings for your board here:
https://github.com/MrChromebox/coreboot/blob/2022.10.24/src/mainboard/google/beltino/variants/tricky/include/variant/thermal.h

user8446, 2022-12-03:
Thank you for the information. 

So I tried `modprobe it87` and I get:

`modprobe: ERROR: could not insert 'it87': Device or resource busy`

dmesg:

```
[ 1558.649919] it87: Found IT8772E chip at 0x700, revision 2
[ 1558.649945] it87: Beeping is supported
[ 1558.649958] ACPI Warning: SystemIO range 0x0000000000000705-0x0000000000000706 conflicts with OpRegion 0x0000000000000700-0x0000000000000706 (\_SB.PCI0.LPCB.SIO.ENVC.ECAP) (20210730/utaddress-204)
[ 1558.649968] ACPI: OSL: Resource conflict; ACPI support missing from driver?
```

```
~$ sudo dmesg | grep -i acpi
[    0.000000] BIOS-e820: [mem 0x000000007e364000-0x000000007e364fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007f3a3000-0x000000007f3a5fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007f3ba000-0x000000007f3bcfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007f6ff000-0x000000007f702fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007f703000-0x000000007f70afff] ACPI data
[    0.000000] efi: SMBIOS=0x7f6e2000 SMBIOS 3.0=0x7f6e0000 ACPI=0x7f70a000 ACPI 2.0=0x7f70a014 MEMATTR=0x7e361018 MOKvar=0x7e34f000 
[    0.019218] ACPI: Early table checksum verification disabled
[    0.019224] ACPI: RSDP 0x000000007F70A014 000024 (v02 COREv4)
[    0.019232] ACPI: XSDT 0x000000007F7090E8 000064 (v01 COREv4 COREBOOT 00000000      01000013)
[    0.019243] ACPI: FACP 0x000000007F708000 000114 (v06 COREv4 COREBOOT 00000000 CORE 20220331)
[    0.019256] ACPI: DSDT 0x000000007F704000 0034B6 (v02 COREv4 COREBOOT 20110725 INTL 20220331)
[    0.019263] ACPI: FACS 0x000000007F724240 000040
[    0.019269] ACPI: SSDT 0x000000007F3BB000 00104F (v02 COREv4 COREBOOT 0000002A CORE 20220331)
[    0.019276] ACPI: MCFG 0x000000007F703000 00003C (v01 COREv4 COREBOOT 00000000 CORE 20220331)
[    0.019282] ACPI: TCPA 0x000000007F3BA000 000032 (v02 COREv4 COREBOOT 00000000 CORE 20220331)
[    0.019289] ACPI: APIC 0x000000007F3A5000 00005C (v03 COREv4 COREBOOT 00000000 CORE 20220331)
[    0.019296] ACPI: HPET 0x000000007F3A4000 000038 (v01 COREv4 COREBOOT 00000000 CORE 20220331)
[    0.019302] ACPI: SSDT 0x000000007F3A3000 00005C (v02 COREv4 SERIALIO 0000002B CORE 20220331)
[    0.019309] ACPI: BGRT 0x000000007E364000 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.019315] ACPI: Reserving FACP table memory at [mem 0x7f708000-0x7f708113]
[    0.019318] ACPI: Reserving DSDT table memory at [mem 0x7f704000-0x7f7074b5]
[    0.019320] ACPI: Reserving FACS table memory at [mem 0x7f724240-0x7f72427f]
[    0.019322] ACPI: Reserving SSDT table memory at [mem 0x7f3bb000-0x7f3bc04e]
[    0.019324] ACPI: Reserving MCFG table memory at [mem 0x7f703000-0x7f70303b]
[    0.019326] ACPI: Reserving TCPA table memory at [mem 0x7f3ba000-0x7f3ba031]
[    0.019329] ACPI: Reserving APIC table memory at [mem 0x7f3a5000-0x7f3a505b]
[    0.019331] ACPI: Reserving HPET table memory at [mem 0x7f3a4000-0x7f3a4037]
[    0.019333] ACPI: Reserving SSDT table memory at [mem 0x7f3a3000-0x7f3a305b]
[    0.019335] ACPI: Reserving BGRT table memory at [mem 0x7e364000-0x7e364037]
[    0.090442] ACPI: PM-Timer IO Port: 0x1008
[    0.090472] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.090476] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
[    0.090484] ACPI: Using ACPI (MADT) for SMP configuration information
[    0.090487] ACPI: HPET id: 0x8086a701 base: 0xfed00000
[    0.226548] ACPI: Core revision 20210730
[    0.233187] ACPI: PM: Registering ACPI NVS region [mem 0x7f6ff000-0x7f702fff] (16384 bytes)
[    0.236292] ACPI: bus type PCI registered
[    0.236295] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
[    0.239406] ACPI: Added _OSI(Module Device)
[    0.239406] ACPI: Added _OSI(Processor Device)
[    0.239406] ACPI: Added _OSI(3.0 _SCP Extensions)
[    0.239406] ACPI: Added _OSI(Processor Aggregator Device)
[    0.239406] ACPI: Added _OSI(Linux-Dell-Video)
[    0.239406] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
[    0.239406] ACPI: Added _OSI(Linux-HPI-Hybrid-Graphics)
[    0.246007] ACPI: 3 ACPI AML tables successfully acquired and loaded
[    0.247711] ACPI: Interpreter enabled
[    0.247738] ACPI: PM: (supports S0 S3 S4 S5)
[    0.247740] ACPI: Using IOAPIC for interrupt routing
[    0.248234] PCI: MMCONFIG at [mem 0xf0000000-0xf3ffffff] reserved in ACPI motherboard resources
[    0.248251] PCI: Using host bridge windows from ACPI; if necessary, use "pci=nocrs" and report a bug
[    0.255843] ACPI: PM: Power Resource [FNP0]
[    0.255877] ACPI: PM: Power Resource [FNP1]
[... 9 lines cut ...]
[    0.264279] ACPI: PCI: Interrupt link LNKB configured for IRQ 0
[    0.264281] ACPI: PCI: Interrupt link LNKB disabled
[    0.264379] ACPI: PCI: Interrupt link LNKC configured for IRQ 0
[    0.264381] ACPI: PCI: Interrupt link LNKC disabled
[    0.264478] ACPI: PCI: Interrupt link LNKD configured for IRQ 0
[    0.264480] ACPI: PCI: Interrupt link LNKD disabled
[    0.264576] ACPI: PCI: Interrupt link LNKE configured for IRQ 0
[    0.264578] ACPI: PCI: Interrupt link LNKE disabled
[    0.264674] ACPI: PCI: Interrupt link LNKF configured for IRQ 0
[    0.264676] ACPI: PCI: Interrupt link LNKF disabled
[    0.264772] ACPI: PCI: Interrupt link LNKG configured for IRQ 0
[    0.264774] ACPI: PCI: Interrupt link LNKG disabled
[    0.264869] ACPI: PCI: Interrupt link LNKH configured for IRQ 0
[    0.264871] ACPI: PCI: Interrupt link LNKH disabled
[    0.266629] ACPI: bus type USB registered
[    0.267342] PCI: Using ACPI for IRQ routing
[    0.291516] pnp: PnP ACPI init
[    0.293175] pnp: PnP ACPI: found 9 devices
[    0.299297] clocksource: acpi_pm: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 2085701024 ns
[    0.337192] ACPI: button: Power Button [PWRF]
[    0.339223] ACPI: thermal: Thermal Zone [THRM] (42 C)
[    0.362903] hpet_acpi_add: no address or irqs in _CRS
```

Any suggestions to get it87 working?

user8446, 2022-12-04:
Ok so the issue is an old and bugged it87 driver that is still in the [kernel](https://docs.kernel.org/hwmon/it87.html). The solution is a patched it87 driver from this [repo]([gaman](https://github.com/gamanakis))

Modeled after [this](https://github.com/lm-sensors/lm-sensors/issues/154#issuecomment-1210022521), here is how to update with the patched it87 driver on the Chromebox (which allows the `ignore_resources_conflict=1` switch for the resources conflict error that you get with the one the kernel):

`git clone git@github.com:gamanakis/it87.git`
`cd it87`
`git switch it8688E`
`make`
`sudo make install`
`sudo modprobe it87 ignore_resource_conflict=1`
`sudo sensors-detect` (it adds the modules for you in `/etc/modules`)
`sensors`
In:
`/etc/modprobe.d/blacklist.conf` (or your own)
Add:
`options it87 ignore_resource_conflict=1`
`sudo update-initramfs -u`
`reboot`

For fan control I went with `fancontrol`. I ran `sudo pwmconfig` to get a baseline and rough adjustments, then fine tuned through `/etc/fancontrol` although you can do it through `pwmconfig` as well.

I tweaked it so I now I have a faster base fan speed of around 3400 RPM which is still quiet and ramps up nicely as temps increase, keeping the box cooler and will prevent overheating.

closed 2022-12-04 by user8446

MrChromebox, 2022-12-04:
I'm happy to adjust the fan curves if you feel they are suboptimal - they're already tweaked from stock slightly

user8446, 2022-12-04:
Hi yes I would increase the fan speed earlier on the temperature curve.
These boxes don't have any passive cooling because of their size. The only way to keep them from melting down is with the fan.

MrChromebox, 2022-12-04:
open to suggested temp/speed combos to adjust as per the linked file above

user8446, 2022-12-05:
Here is the curve that [fancontrol](https://linux.die.net/man/8/fancontrol) is calculating. With these fan speeds running `stress -c 2` the temps would never go over 46c and adding videos to tax the GPU at the same time I couldn't get it over 50c so in my opinion this is a perfect balance for temps and noise. The sensor low end starts at 30c

```
30c -> 2600prm
37c -> 2667rpm
40c -> 3040rpm
41c -> 3139rpm
43c -> 3250rpm
44c -> 3391rpm
45c -> 3450rpm
46c -> 3650rpm
47c -> 3792rpm
48c -> 3857rpm
49c -> 4041rpm
50c -> 4115rpm
```

On this box the fan was `hwmon2/fan2` and the control was `hwmon2/pwm2`.  The ACPI interface temp and the "package ID0" temp were always pretty much in lockstep so I used that instead of an individual core temp when setting up fancontrol

```
~$ sensors
it8772-isa-0700
Adapter: ISA adapter
in0:           2.22 V  (min =  +2.68 V, max =  +2.86 V)  ALARM
in1:           2.22 V  (min =  +2.27 V, max =  +2.15 V)  ALARM
in2:           2.04 V  (min =  +1.79 V, max =  +2.68 V)
+3.3V:         3.36 V  (min =  +6.02 V, max =  +4.03 V)  ALARM
in4:           2.22 V  (min =  +2.57 V, max =  +2.66 V)  ALARM
in5:           2.22 V  (min =  +0.65 V, max =  +0.78 V)  ALARM
in6:           2.22 V  (min =  +0.00 V, max =  +2.84 V)
3VSB:          3.34 V  (min =  +5.93 V, max =  +5.35 V)  ALARM
Vbat:          3.24 V  
fan2:        3199 RPM  (min =   55 RPM)
temp1:       -128.0°C  (low  = +15.0°C, high = +84.0°C)
temp2:       -128.0°C  (low  = -109.0°C, high = +124.0°C)
temp3:        -60.0°C  (low  = -97.0°C, high = -73.0°C)  ALARM  sensor = Intel PECI
intrusion0:  ALARM

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +41.0°C  (crit = +98.0°C)

pch_haswell-virtual-0
Adapter: Virtual device
temp1:        +91.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +36.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +35.0°C  (high = +100.0°C, crit = +100.0°C) 
```

MrChromebox, 2022-12-05:
any chance you are able to provide the duty cycle value for 3k/4k RPMs? the fans on these are all slightly different (non-linear) so I can't use Panther as a baseline for Tricky

MrChromebox, 2022-12-06:
@user8446 drop me an email (mrchromebox at gmail), will have you try a test build and see if you still need the manual override

user8446, 2022-12-06:
Sent.