MrChromebox/firmware #411: IP Addressing oddity -probable BUG-

markosjal, 2022-12-05:
I have isolated this issue now that I may or may not have reported here or elsewhere in one form or another, but now I can Make it happen on an Asus CN60 "at will" and Can NOT reproduce it on other hardware such as Mission One Mini PC. The trick is when I started testing with a second CN60 chrome box and was able to get the same results with a few tweaks it told me more clearly how to reproduce it.

Hardware....
Chromebox CN60 4GB RAM internal SSD

Procedure (exactly what I have done to reproduce the error and not all steps may be necessary) .......
- Install MrCromebox for UEFI booting
- Install Libreelec to internal SSD
- Boot to Libreelec 
- Install and switch to Confluence skin (may not be required but I did this on all test systems because it is easier for me) 
- In Libreelec add-on set a fixed IP address (I used 10.0.0.100 and 10.0.0.101 with identical results. I tested this mostly with 5Ghz Wifi but saw the same results with Ethernet and 2.4 Ghz Wifi)
- Set IP V6 to Auto
- Set DNS Server to 1.1.1.1 and 1.0.0.1 (may not be necessary but as I did)
- Make sure network is set to auto connect if WiFi as there is a bug in Libreelec and it may not retain this setting at first. You may need to set again after reboot
- Make Sure Firewall is disabled in. Libreelec settings
- reboot Libreelec
- Make sure Libreelec connects to Wireless if using wireless, and if not change the auto connect setting and reboot again
- Now with Libreelec running and a fixed IP address go to a computer on the LAN and initiate a Ping
- When Libreelec first starts there will probably be no issue, but after several minutes you will probably start to see that the ping stops getting replies from Libreelec box
- sometimes at this point if you go and create some activity on Libreelec the ping MAY start getting replied to again , but not always.

So what this means is that this MrChromebox CN60 Libreelec box with the IPTV Merge Add-on with web server enabled (or a plethora of other addons) the box becomes inaccessible as it also becomes unavailable via ssh, as a web server , etc.

It however does not seem to affect outbound connections, for instance while I can not Ping Libreelec I can access my NAS from Libreelec and can access internet based sources like Hulu, Netflix, Paramount+, Frndly TV , etc from Libreelec

I have now confirmed this on a SECOND Asus CN60 and can CONSISTENTLY reproduce the same issue. 

I have also removed the internal SSD and installed to an SD card and got the same results. In fact that same SD card will NOT give me the same issue when I use it to boot the Mission One Mini PC but will CONSISTENTLY give me the same results on two different CN60s

I am unable to reproduce similar results on a Mission One Mini PC with intel hardware and exactly the same Wireless card. 

I am reporting this as a possible bug on Libreelec forums too  as we do not know if it is a Libreelec issue or otherwise.

I also want to note that during this diagnostic process..... 
I originally tore apart my entire network infrastructure and replaced it ALL, thinking I had a switch or wireless access point failing. NONE of my network hardware; switch, router, access point remains the same but the problem on the CN60 does and also remains even when connected to Ethernet!

MrChromebox, 2022-12-05:
I'm honestly not sure how this could be in any way firmware related; will be curious to see if any of the LE devs have any insight. Does the issue persist across different LE/kernel versions?

markosjal, 2022-12-06:
I have seen the same on LE versions 10.0.2 and 10.0.3 , They are the only
ones I have tested . I figure `after putting months into being able to
identify this then reproduce this now at will I have done more than my
share.

Unfortunately I am giving the other chromebox back to my brother in law and
will not have access to the second chromebox to test this.

At least I now know I can not use the chromebox as a "server" of any kind
with LE due to this bug, whatever its cause is.

MrChromebox, 2022-12-06:
I haven't used LE in years so I can't really say, but haven't seen any issues with the CN60 running ubuntu server or anything similar

markosjal, 2022-12-06:
It is very easy to test and I am  certain it will show up as I indictated
in my post. That is precisely why I documented it. Where the problem comes
from however is beyond my ability to say. It is interesting to note that
the problem does not happen on Mission One mini PC but happens on two
different cn60s . Chipset support?? It would be interesting to see if the
same happens on other chromeboxes.

markosjal, 2022-12-06:
I am wondering if there is a MiniPC that has identical hardware  to the CN60 that someone might test. Might help to rule out MrChromebox

MrChromebox, 2022-12-06:
does the issue exist when using DHCP? Just assign a static address based on MAC on your router

markosjal, 2022-12-06:
What I might call "micro failures" seem to occur on DHCP, where a one
minute ping may lose a few packets, the problem is not nearly as
severe with DHCP

I am not looking for that kind of work-around, I am looking to see if it
can be made to function correctly by finding the cause of the problem.

Again not looking for a work-around of DHCP which I have been aware of for
a while now

markosjal, 2022-12-06:
> I understand, just trying to understand the scope of the issue. Given what
> functions the firmware is performing for hw init, I don't see how this can
> be a firmware issue, unless it's related to PCIe power/performance.
>
> Can you do lspci -vvv for the network controllers on both boxes and see
> if any differences?

My initial reaction when I discovered how this bug manifested itself was to
investigate any power saving modes in Kodi and Libreelec, which there
really appear to be none.

Power saving was my first thought as it seems to happen sometime after the
system has ran for a while and sometimes works normally again after there
is some local activity (or that could be coincidence)

Oddly enough it happens on internal Ethernet or WIfi which also made me
think of a power save mode.

markosjal, 2022-12-21:
I have now also confirmed this same problem on Aus CN62 chromebox!

MrChromebox, 2022-12-21:
well, the CN60 and CN62  are pretty much the exact same board, with the same LAN NIC, just (ever so slightly) different SoC

closed 2026-11-04 by MrChromebox