MrChromebox/firmware #412: Asus Chromebook C425 mrchromebox Full ROM UEFI, Ubuntu 22.10, audio not working

xz3f2w500, 2022-12-10:
Asus Chromebook C425 mrchromebox Full ROM UEFI, Ubuntu 22.10, audio not working

bqv, 2022-12-10:
Me too, on a C433 SHYVANA - added chromeos firmware manually but no dice

Can start a new issue for it if you want though @MrChromebox

bqv, 2022-12-11:
Boot log at my state [here](https://hardbin.com/ipfs/QmUEb1UjC6abN19rohVWFwHvvHEEienKLw3PoM6ajPnBwq/#8z9UBEuphNeQbFEYLt5aDUzuMiBwGm8KpcFZdHtNjrAq) 

I moved snd_hda_intel to initramfs, in case confusing

MrChromebox, 2022-12-17:
this isn't a firmware issue, nothing I can do that Intel doesn't support SKL/KBL Chromebook audio under Linux currently

closed 2022-12-17 by MrChromebox

psipika, 2023-01-17:
> this isn't a firmware issue, nothing I can do that Intel doesn't support SKL/KBL Chromebook audio under Linux currently

@MrChromebox would you be kind enough to elaborate on this, please? I see [`max98927.c`](https://github.com/torvalds/linux/blob/master/sound/soc/codecs/max98927.c) in the kernel source and configuration shows it as a module. I was hoping you could provide a bit more insight into the lack of support you mention.

MrChromebox, 2023-01-17:
@psipika there's more than just the codec itself, needs to be support for the SoC's DSP, loading the firmware, and then the topology for board in question. Some of the folks in the chrultrabook Discord are working on it currently. But there's nothing I can do from the firmware side to make it magically work - it's a Linux kernel/audio subsystem issue.

psipika, 2023-01-17:
> @psipika there's more than just the codec itself, needs to be support for the SoC's DSP, loading the firmware, and then the topology for board in question. Some of the folks in the chrultrabook [...]

I did not expect the issue to be firmware-related, as you'd mentioned earlier on in this issue. I appreciate the additional details -- thank you!

psipika, 2023-01-27:
@xz3f2w500 and @bqv I got my sound to work on my headphones thanks to info I got on chrultrabook discord. I ended up cloning the chromeos kernel (5.15, same version on my 22.04 system), configuring it based on [EupneaOS](https://github.com/eupnea-linux) and running [audio-scripts](https://github.com/eupnea-linux/audio-scripts)

DO THIS **ONLY** IF YOU:
- FEEL COMFORTABLE BUILDING YOUR OWN KERNEL
**AND**
- KNOW HOW TO GET BACK TO A PREVIOUS (working) KERNEL VERSION:

1. `git clone --depth 1 -b chromeos-5.15 https://chromium.googlesource.com/chromiumos/third_party/kernel kernel-5.15`
2. Copy [this] into `kernel-5.15/.config`
3. Inside `kernel-5.15` run `make`, then `sudo make INSTALL_MOD_STRIP=1 modules_install`, then `sudo make install`
4. Reboot into the new kernel.
5. .`git clone https://github.com/eupnea-linux/audio-scripts.git` 
6. Inside `audio-scripts` run `./setup-audio`
7. Reboot.

cgioconda, 2023-02-10:
What is [this]?
> 2. Copy [this] into `kernel-5.15/.config`

psipika, 2023-02-10:
> What is [this]?
> 
> > 2. Copy [this] into `kernel-5.15/.config`

Oops: https://gist.github.com/psipika/94b1fb034076a06380380e6b3c9d5445#file-chromeos_kernel_config-5-15

cgioconda, 2023-02-14:
> @xz3f2w500 and @bqv I got my sound to work on my headphones thanks to info I got on chrultrabook discord. I ended up cloning the chromeos kernel (5.15, same version on my 22.04 system), configuring [...]

I was told that MAX98927 is supported in kernel 5.10. Is it possible for you to generate a compatible chromeos kernel config file for 5.10, and share that so I can test?

bqv, 2023-02-14:
I was the first one to get that working, with the help of that discord (before they banned me for being British :|)

I have it running on a 5.10 kernel

cgioconda, 2023-02-14:
> I was the first one to get that working, with the help of that discord (before they banned me for being British :|)
> 
> I have it running on a 5.10 kernel

Are you able to provide some documentation on how you accomplished this? Very interested to see.

bqv, 2023-02-14:
They gave me a prebuilt kernel, just ask them for that

psipika, 2023-02-14:
> They gave me a prebuilt kernel, just ask them for that

Follow these directions: https://github.com/olm3ca/Pixel-Slate#part-3-linux-and-audio

psipika, 2023-03-18:
@MrChromebox I've just updated to the latest (20230123) version of the LEONA firmware, running 5.15.103 chromeos kernel and latest audio-scripts. I've got the audio set-up, but the latest audio-scripts appears to offer the opportunity to get the speakers working (via avs). I ran the `setup-audio` script, and `dmesg` is reporting something like this:

```bash
[Sat Mar 18 14:26:56 2023]  Digital Microphone: ASoC: error at dpcm_fe_dai_hw_params on Digital Microphone: -2
[Sat Mar 18 14:26:56 2023] snd_soc_avs 0000:00:1f.3: no DMIC ep_blob found
[Sat Mar 18 14:26:56 2023] snd_soc_avs 0000:00:1f.3: module-type create failed: -2
[Sat Mar 18 14:26:56 2023] snd_soc_avs 0000:00:1f.3: error creating module -2
[Sat Mar 18 14:26:56 2023] snd_soc_avs 0000:00:1f.3: create path failed: -2
[Sat Mar 18 14:26:56 2023] snd_soc_avs 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on DMIC Pin: -2
[Sat Mar 18 14:26:56 2023]  DMIC: ASoC: error at __soc_pcm_hw_params on DMIC: -2
```
In the discord, I was told I should upgrade coreboot:
```
you need to compile coreboot with 4ch dmic blobs
```
I'm wondering if the coreboot installed via your firmware update script has the `4ch dmic blobs` enabled?