MrChromebox/firmware #413: Chromebook Pixel 2013 with ChromeOS Flex (Cloudready) Fan runs on full power all the time

gsstark, 2022-12-13:
Not sure if this is the same as #200 or a different issue. Freshly flashed 2013 Pixel with freshly installed and updated Chromeos Flex. The fan is running at full speed all the time :(

Firmware is 4.18.1 1/27/2022. CPU is i5-3427U @ 1.8Ghz. The fan doesn't run in the bios screen, only when the Chromeos Flex logo appears. And in fact if I go to the bios device settings screen and then hit reboot it seems the fan is disabled (or at least doesn't come on right away) when it boots.

MrChromebox, 2022-12-13:
this sounds like a CrOS Flex issue not a firmware one. Could be something with the EC driver (or lack thereof)? I've not tested Flex on LINK, but have no such issues running Linux on mine

elektroencefalografista, 2023-01-29:
Can confirm the same issue on mine. Fan works fine in Linux and brunch ChromeOS 96, but runs at full tilt in ChromeOS Flex.

MrChromebox, 2023-02-03:
dupe of #394

closed 2023-02-03 by MrChromebox