MrChromebox/firmware #415: flashrom segfaults when trying to upgrade nightfury from 4.18.1 to 4.18.2

mowgli80, 2022-12-18:
OS: Ubuntu 22.10

Kernel: 5.19, 5.15

[  958.557840] flashrom[5289]: segfault at 400000007060 ip 00007faf580661ce sp 00007ffdc41e5a40 error 4 in libpci.so.3.8.0[7faf58063000+9000]
[  958.557849] Code: 24 08 31 c0 89 54 24 04 40 f6 c6 03 75 4d 8d 46 04 3b 87 88 01 00 00 7e 33 48 8b 87 78 01 00 00 48 8d 54 24 04 b9 04 00 00 00 <ff> 50 48 48 8b 54 24 08 64 48 2b 14 25 28 00 00 00 75 2f 48 83 c4

MrChromebox, 2022-12-18:
you're using a distro with a newer libpci than flashrom was compiled against. I don't have a good solution at the moment to handle this sort of breakage. Compiling flashrom on the device itself is a bit of a hassle

MrChromebox, 2023-02-03:
I just updated the script to use a newer version of flashrom, along with some other changes. LMK if that fixes the issue for you

mowgli80, 2023-02-03:
It worked. Thanks for the fix.

closed 2023-02-03 by mowgli80