MrChromebox/firmware #416: Firmware-util.sh suddenly not working

tlscott9999, 2022-12-24:
Our non-profit organization uses the MrChromebox script to prepare EOL chromebooks to use Linux.  These chromebooks are then distributed free to local students and families.

This morning while finishing up a batch of identical Dell Chromebook 11 P22t units, the Firmware-util.sh is doing something wonky and won't let me complete the process.  Over the past 2 days I've done 100+ identical units without problem.

The problem comes when I choose #2 from the menu.  I answer Y to the first 2 questions and Usually answer N to the last question to backup ChromeOS.  Now I do not get the question to answer, but a statement that I must backup:

"Connect the USB/SD device to store the firmware backup and press [Enter] to continue................"

When do press enter I get a message "No USB devices available to store firmware backup" in red.  I have tried both having a USB drive plugged and not having a USB plugged in -- both with same response.  I have tried the process on multiple identical units and had the same response.

We go through a few hundred devices a month and I usually only make a backup once for all similar devices.

Any suggestions?

[image]

MrChromebox, 2022-12-24:
no idea why it wouldn't be detecting your USB device, as there's been no change there in ages. But I did just push a change that will allow you to skip the backup, which is completely unnecessary on EOL devices like WOLF.

iamkevin217, 2023-01-22:
@mrChromebox where can I find this fix? I'm having a similar issue. I'm running bare bones canary 67 and it ins't detecting my USB when i try to install the firmware; nor will it let me bypass it.

MrChromebox, 2023-01-22:
@iamkevin217 there is nothing to find, I changed the script. What device are you using?

tlscott9999, 2023-02-03:
@MrChromebox/firmware ***@***.***> - sorry to bother you
again, but a new error popped up today on some HP Chromebook 11 G3/G4 while
running the script.  I have done over 200 identical systems over the past
week with no errors until today.  Below is a photo of the problem which
occurs near the bottom of the script.  cat: /tmp/flashrom, log: No such
file or directory An error occurred flashing the Full ROM firmware.

Please help when you get a chance.  Thanks.

[image: image.png]

Terry Scott
Director
2nd Chance Computer Refurbishing Modesto
209-402-8586
***@***.***
www.2ndChanceModesto.org
- Providing students with computers at no cost -

MrChromebox, 2023-02-03:
@tlscott9999 I pushed a fix, LMK if that takes care of it

iamkevin217, 2023-02-03:
Lenovo 500e. It ended up working after the third try or so. Thanks.

tlscott9999, 2023-02-03:
@MrChromebox/firmware ***@***.***> - unfortunately no, I
have a new red error message at the bottom of the screen now.

/usr/sbin/flashrom: unrecognized option --noverify-all' Please run
"flashrom --help" for usage info..
J Error running cmd: /usr/sbin/flashrom -p internal --noverify-all -w
coreboot_tiano-kip-mrchromebox_20230129.rom >/dev/null 2>&1 -o
/tmp/flashrom, le
Press enter to view the flashrom log file, then space for next page, q to
quit

[image: image.png]

Terry Scott
Director
2nd Chance Computer Refurbishing Modesto
209-402-8586
***@***.***
www.2ndChanceModesto.org
- Providing students with computers at no cost -

MrChromebox, 2023-02-03:
ok, should be fixed for reals now

tlscott9999, 2023-02-03:
Thanks. I will try it when I get home.

--Terry Scott
sent from my Android device

tlscott9999, 2023-02-03:
@MrChromebox/firmware ***@***.***> sorry, I'm still
getting a similar message.

 Error running cmd: /usr/sbin/flashrom -p internal --noverify-all -w
coreboot_tiano-kip-mrchromebox 20230129.rom >/dev/null 2>&1 -o
2tmp/flashrom. Log An error occurred flashing the Full ROM firmware. DO NOT
REBOOT!

[image: image.png]

Terry Scott
Director
2nd Chance Computer Refurbishing Modesto
209-402-8586
***@***.***
www.2ndChanceModesto.org
- Providing students with computers at no cost -

MattDeVillier-AMD, 2023-02-03:
your images are not getting attached, need to drag into the comment IIRC

the flashrom command looks sane, but it's not producing a log file which means it't not erasing/writing anything. Not sure if this is a flashrom command issue or a bash variable one

tlscott9999, 2023-02-03:
[image]

Let me know if the image is not uploaded now.

MrChromebox, 2023-02-03:
please run the script as: sudo bash -x firmware-util.sh
that will show all the bash processing. Navigate script as before, then take pic when you get the error again. Need to see how bash is processing variables

tlscott9999, 2023-02-03:
[image]
Here you go.

MrChromebox, 2023-02-03:
ok, found/fixed I hope. was a bug in ChromeOS's flashrom =/

tlscott9999, 2023-02-04:
Yes, that works!  Thanks so much for your help.

--Terry Scott
sent from my Android device

MrChromebox, 2023-02-04:
closing as fixed

closed 2023-02-04 by MrChromebox

Timmyraysmom, 2024-07-18:
I'm trying yo run firmware script in my chromebook wp disabled try to run command line and get this message: Warning: Failed to open the file firmware-util.sh: Read-only file system

curl: (23) Failure writing output to destination

MrChromebox, 2024-07-18:
> I'm trying yo run firmware script in my chromebook wp disabled try to run command line and get this message: Warning: Failed to open the file firmware-util.sh: Read-only file system
> 
> curl: (23) Failure writing output to destination

most likely scenario: you're not following directions and have logged in or are trying to run the script command as root.

Timmyraysmom, 2024-07-22:
I'm teaching myself about computers and I thought I was following them everytime maybe I just don't get it or maybe I shouldn't try