MrChromebox/firmware #417: ASUS Chromebook C523NA Slow Charging (USB TYPE C)

idontknowanything01, 2022-12-27:
I am currently using coreboot 4.18.2 UFEI firmware
My ASUS laptop comes with a 45W fast charger using type c interface for charging it taking 3-4 hours to complete the charging but in unmodified bios it took only 45 min to an hour.

I restored the default firmware and found the hardware is fine and charging is happening at the speed before firmware modification.

A few more details
1) In your firmware once i connect the charger the charging indicator change its colour after 5 seconds.
2) My OS (Linux Mint) show the charging indicator after 30 Sec or so.
3) Once i disconnect the charger the charger the charging indicator change its colour instantly.
4)  Once i disconnect the charger the charger My OS stop showing the charging indicator after a min or so.

MrChromebox, 2022-12-27:
OS driver is responsible for negotiating charging speed, make sure the cros-ec driver is loaded/working

idontknowanything01, 2022-12-27:
Thanks a lot for the replay can you please elaborate a little bit about how can i install cros-ec or check its installed or not any how to link will also work thanks again.

lee321987, 2023-12-19:
Not a solution, but a bit of a workaround:  On my Asus Chromebook C423, i can get fast charging if I completely shut down the laptop, connect the charger and boot up, but if the cord is unplugged and re-plugged in it goes back to super-slow charging.
Did anyone ever find a solution to this?

ChocolateLoverRaj, 2024-03-09:
Some useful commands for debugging slow charging:
```
sudo ectool usbpdpower
```
```
04:25:09 rajas@Redrix ~ → sudo ectool usbpdpower
Port 0: SNK Charger PD 19680mV / 3000mA, max 20000mV / 3000mA / 60000mW
Port 1: SRC 3000mA
```
```
sudo ectool typecstatus <port number>
```
```
04:25:11 rajas@Redrix ~ → sudo ectool typecstatus 0
Port C0: enabled, connected  State:Attached.SNK
Role:SNK DFP VCONN, Polarity:CC1
CC State: UNKNOWN
MUX: USB=1 DP=0 POLARITY=NORMAL HPD_IRQ=0 HPD_LVL=0
     SAFE=0 TBT=0 USB4=0
Port events: 0x00000000
SOP  PD Rev: 3.0
Source Capabilities:
    Fixed: 5000mV 3000mA UP DRD
    Fixed: 9000mV 3000mA 
    Fixed: 15000mV 3000mA 
    Fixed: 20000mV 3250mA 
```

cmbuck, 2024-06-20:
I'm observing the same issue on two different Asus C302 devices.  On ChromeOS the devices appear to charge normally, as expected (full charge from ~40% in <1hr).  After flashing new firmware and installing Linux, both devices charge very slowly.

Additional details and troubleshooting steps below:
* I have two Asus C302 (Cave) devices: Device A running PopOS 22.04 and Device B running ChromeOS
* Device A exhibits slow charging (ETA of 10 hours to charge from ~40% battery SoC with 65W USB-PD charger)
* Purchase new aftermarket battery, Device A still exhibits slow charging even with brand-new aftermarket battery
* Insert Device A's OEM battery into Device B which is running ChromeOS.  Battery charges quickly (40%->100% ETA approximately 40mins) **Failure follows the device, not the battery**
* Flash MrChromebox firmware onto Deice B following instructions from chultrabook.com.  Install Ubuntu 24.04 on Device B
* Device B now exhibits slow charging failure identical to Device A running PopOS 22.04

```
christopher@christopher-Cave2:~/Downloads$ sudo ./ectool usbpdpower
Port 0: SNK Charger PD 19570mV / 2248mA, max 20000mV / 2250mA / 45000mW
Port 1: Disconnected
christopher@christopher-Cave2:~/Downloads$ sudo ./ectool typecstatus 0
EC result 1 (INVALID_COMMAND)
Port C0: enabled, connected  State:SNK_READY
Role:SNK DFP, Polarity:CC1
PD Partner Capabilities:
 DR data
 Unconstrained power
```

`upower` indicates the battery is charging at ~2.5W which is really low.
```
christopher@christopher-Cave2:~/Downloads$ upower -i /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               AS19MZF
  model:                C302-50
  serial:               05DD
  power supply:         yes
  updated:              Wed 19 Jun 2024 08:49:07 PM PDT (15 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              21.5754 Wh
    energy-empty:        0 Wh
    energy-full:         28.8057 Wh
    energy-full-design:  39.0005 Wh
    energy-rate:         2.5333 W
    voltage:             8.13 V
    charge-cycles:       164
    time to full:        2.9 hours
    percentage:          74%
    capacity:            73.8598%
    technology:          lithium-ion
    icon-name:          'battery-full-charging-symbolic'
  History (rate):
    1718855317	2.533	charging
    1718855287	2.564	charging
    1718855257	2.541	charging

```

I was originally concerned that this was a hardware issue, but the sequence of steps I noted above strongly indicates a software cause.

Is there a way to investigate how much power the battery is requesting, or how much power the EC is instructing the battery charger to pass to the battery?

MrChromebox, 2024-06-20:
sounds like ChromeOS was doing something with the PD controller to adjust the charge rate that Linux isn't

shiptux, 2024-07-08:
Is there any helpful of acpi_call?

shiptux, 2024-08-02:
I have the same issue on redrix,  howerver after upgrade my kernel to 6.6.43-chrultrabook, everything of charging works fine, include rapid charge.

cmbuck, 2024-08-10:
@shiptux  Glad to hear you've found a resolution.  I tried to replicate your success but I'm still observing my charge rate being limited to 2.5W on the latest chultrabook kernel.

```
christopher@pop-os:~$ uname -r
6.6.43-chrultrabook
christopher@pop-os:~$ upower -i /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               AS19MZF
  model:                C302-50
  serial:               091B
  power supply:         yes
  updated:              Fri 09 Aug 2024 11:23:20 PM PDT (119 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              32.7864 Wh
    energy-empty:        0 Wh
    energy-full:         37.8632 Wh
    energy-full-design:  38.494 Wh
    energy-rate:         2.4168 W
    voltage:             8.35 V
    charge-cycles:       18
    time to full:        2.1 hours
    percentage:          86%
    capacity:            98.3613%
    technology:          lithium-ion
    icon-name:          'battery-full-charging-symbolic'

```

In order to install the custom kernel, I took the steps:
* Clone [chultrabook kernel repo](https://github.com/chrultrabook/debian-kernel)
* run `./build.sh`
* install the resulting deb packages `sudo dpkg -i *.deb`
* Manually edit my grub config to boot the new kernel

Did I miss any steps?  I'm still relying on @lee321987's workaround whenever I need to charge the device faster than 2.5W.

shiptux, 2024-08-10:
> @shiptux Glad to hear you've found a resolution. I tried to replicate your success but I'm still observing my charge rate being limited to 2.5W on the latest chultrabook kernel.
>
> ``` [...]

I don't install the kernel manually, actually I installed it by run the https://github.com/WeirdTreeThing/chromebook-linux-audio scripts. I will try your steps.

```bash
shiptux@shiptux-redrix:~$ upower -i /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               333-2C-
  model:                DK04050
  serial:               13C8
  power supply:         yes
  updated:              2024年08月10日 星期六 15时08分14秒 (27 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              26.9654 Wh
    energy-empty:        0 Wh
    energy-full:         44.044 Wh
    energy-full-design:  50.974 Wh
    energy-rate:         35.0812 W
    voltage:             8.281 V
    charge-cycles:       46
    time to full:        29.2 minutes
    percentage:          61%
    capacity:            86.4048%
    technology:          lithium-ion
    icon-name:          'battery-full-charging-symbolic'
  History (charge):
    1723273694	61.000	charging
  History (rate):
    1723273694	35.081	charging
    1723273663	34.588	charging
    1723273662	33.541	charging
    1723273661	33.264	charging
    1723273660	4.173	charging
    1723273659	10.357	discharging
    1723273649	11.935	discharging
    1723273619	4.951	discharging

shiptux@shiptux-redrix:~$ uname -r 
6.6.43-chrultrabook

```

shiptux, 2024-08-14:
> @shiptux Glad to hear you've found a resolution. I tried to replicate your success but I'm still observing my charge rate being limited to 2.5W on the latest chultrabook kernel.
>
> ``` [...]

I am trying to add this patch(https://github.com/chrultrabook/debian-kernel/blob/main/patches/i2c-Add-device-property-for-probing.patch) to my distribution's kernel. Everything works well including the battery charge speed.

lee321987, 2025-03-11:
I just tried a live Puppy Linux USB (BookwormPup64_10.0.10.iso) and it gets full charge speed any time I plug in.